MarlinFirmware/Marlin #21994: [BUG] (bug summary)

brianredbeard, 2021-05-28:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

Somewhere between 22de37ad5037eb829de9e817f74d25768bd8e117 and 0d7075b90c225d2f88e1d790b2df4acbdfcfac8b probing with the BLtouch during `G29` operations has stopped occurring.  

Upon startup, the BLTouch unit itself successfully performs a power on self test.  Subsequently, the probe successfully deploys upon `G28` execution.  Execution of `G29` shows `BLTouch DEPLOY requested` but, as can be seen in the video, this operation does not occur.

### Bug Timeline

> 2.0.8.x

### Expected behavior

Proper deployment of probe

### Actual behavior

Probe deploys during `BLTouch DEPLOY requested`

### Version of Marlin Firmware

0d7075b90c225d2f88e1d790b2df4acbdfcfac8b

### Printer model

Ender 3 Pro

### Electronics

BTT E3 DIP, TMC5160

### Add-ons

BLTouch

### Host Software

OctoPrint

### Additional information & file uploads

### Video of the problem:
https://user-images.githubusercontent.com/57542/120032839-a1d1df00-bfaf-11eb-9b1d-f2a76878bc90.mp4

### Logfiles:
[serial.log](https://github.com/MarlinFirmware/Marlin/files/6562948/serial.log) 

### Configurations:
[Configurations.zip](https://github.com/MarlinFirmware/Marlin/files/6563149/Configurations.zip)

Output of `git diff -w -U0 0d7075b90c225d2f88e1d790b2df4acbdfcfac8b -- Marlin/Configuration{,_adv}.h`:
https://gist.github.com/brianredbeard/015a0c4f8cb4e8625ddb9b1d7b3a506b

brianredbeard, 2021-05-29:
FYI, seems to be some interaction with UBL.

Re-building and swapping `AUTO_BED_LEVELING_UBL` for `AUTO_BED_LEVELING_BILINEAR` has the probe deploying on `G29`.

ldursw, 2021-05-29:
It looks like the same issue as #21429 but the bad commit is not in the range you specified.

GhostlyCrowd, 2021-05-31:
@brianredbeard  

Can you try building the latest bugfix and instead of using the default envs for your board use the STM32 HAL version of STM32F103RC_btt envs. 

`[env:STM32F103RC_stm32]
platform             = ${common_stm32.platform}
extends              = common_stm32
board                = genericSTM32F103RC
monitor_speed        = 115200
board_build.core     = stm32
board_build.variant  = MARLIN_F103Rx
board_build.ldscript = ldscript.ld
extra_scripts        = ${common.extra_scripts}
  pre:buildroot/share/PlatformIO/scripts/generic_create_variant.py
  buildroot/share/PlatformIO/scripts/stm32_bootloader.py

[env:STM32F103RC_btt_stm32]
platform             = ${common_stm32.platform}
extends              = env:STM32F103RC_stm32
build_flags          = ${common_stm32.build_flags} -DDEBUG_LEVEL=0 -DTIMER_SERVO=TIM5
board_build.offset   = 0x7000
board_build.encrypt  = No
board_build.firmware = firmware.bin
board_upload.offset_address = 0x08007000

[env:STM32F103RC_btt_USB_stm32]
extends           = env:STM32F103RC_btt_stm32
platform          = ${common_stm32.platform}
platform_packages = framework-arduinoststm32@https://github.com/rhapsodyv/Arduino_Core_STM32/archive/usb-host-msc-cdc-msc.zip
build_unflags     = ${common_stm32.build_unflags} -DUSBD_USE_CDC
build_flags       = ${env:STM32F103RC_btt_stm32.build_flags} ${env:stm32_flash_drive.build_flags}
  -DUSBCON
  -DUSE_USBHOST_HS
  -DUSBD_IRQ_PRIO=5
  -DUSBD_IRQ_SUBPRIO=6
  -DUSE_USB_HS_IN_FS
  -DUSBD_USE_CDC_MSC

[env:STM32F103RC_btt_512K_stm32]
platform          = ${common_stm32.platform}
extends           = env:STM32F103RC_btt_stm32
board_upload.maximum_size = 524288
build_flags       = ${env:STM32F103RC_btt_stm32.build_flags} -DLD_MAX_DATA_SIZE=524288 -DSTM32_FLASH_SIZE=512

[env:STM32F103RC_btt_512K_USB_stm32]
platform          = ${common_stm32.platform}
extends           = env:STM32F103RC_btt_USB_stm32
board_upload.maximum_size = 524288
build_flags       = ${env:STM32F103RC_btt_USB_stm32.build_flags} -DLD_MAX_DATA_SIZE=524288 -DSTM32_FLASH_SIZE=512
`

brianredbeard, 2021-06-03:
FYI: I am still engaged on this, but waiting on the STM32 HAL switch-over to stabilize (i.e. #21999, #22019, #22022, etc)

closed 2021-08-14 by github-actions[bot]