MarlinFirmware/Marlin #21995: [BUG] (bug summary)

iiKelZenii, 2021-05-29:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

Marlin build error.

Configuration.h:
#define REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER
#define MOTHERBOARD BOARD_RAMPS_14_EFB
[config.tar.gz](https://github.com/MarlinFirmware/Marlin/files/6564151/config.tar.gz)

OS: Ubuntu 20.04
IDE: ArduinoIDE  1.8.13
Libs: u8glib v 1.19.1 or 1.18.0

I reinstalled u8glib but it didn't help.

Build error message:
In file included from /tmp/arduino_build_319465/sketch/src/lcd/dogm/marlinui_DOGM.h:31:0,
                 from /tmp/arduino_build_319465/sketch/src/lcd/marlinui.cpp:126:
/tmp/arduino_build_319465/sketch/src/lcd/dogm/HAL_LCD_class_defines.h: In member function 'void U8GLIB_64128N_2X_HAL::init(pin_t, pin_t, pin_t, pin_t, pin_t)':
/tmp/arduino_build_319465/sketch/src/lcd/dogm/HAL_LCD_class_defines.h:37:13: error: 'init' is not a member of 'U8GLIB'

if git checkout 2.0.7.2 - build ok
if git checkout 2.0.8 - build error (U8GLIB)
if git checkout bugfix-2.0.x build error (U8GLIB) 

### Version of Marlin Firmware

bugfix-2.0.x. commit b80179d14a23911dc81dfa7fe9d6fca653326798

### Your Slicer

Cura

ellensp, 2021-05-29:
duplicate of https://github.com/MarlinFirmware/Marlin/issues/21967

ellensp, 2021-05-29:
duplicate of https://github.com/MarlinFirmware/Marlin/issues/21948

closed 2021-05-29 by ellensp