MarlinFirmware/Marlin #21997: [BUG] (bug summary) compiling Problem version 02000801

arnold-b, 2021-05-29:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

Can not compile latest bugfix
In file included from J:\Marlin-2.0.x\Marlin\src\inc\MarlinConfigPre.h:39:0,
                 from J:\Marlin-2.0.x\Marlin\src\inc\MarlinConfig.h:28,
                 from J:\Marlin-2.0.x\Marlin\src\MarlinCore.h:24,
                 from J:\Marlin-2.0.x\Marlin\src\MarlinCore.cpp:31:
Configuration.h:1:1: error: stray '\357' in program
 ﻿/**
 ^
Configuration.h:1:2: error: stray '\273' in program
 ﻿/**
  ^
Configuration.h:1:3: error: stray '\277' in program
 ﻿/**
   ^
In file included from J:\Marlin-2.0.x\Marlin\src\inc\MarlinConfigPre.h:39:0,
                 from J:\Marlin-2.0.x\Marlin\src\inc\MarlinConfig.h:28,
                 from J:\Marlin-2.0.x\Marlin\src\HAL\AVR\MarlinSerial.cpp:39:
Configuration.h:1:1: error: stray '\357' in program
 ﻿/**
 ^
Configuration.h:1:2: error: stray '\273' in program
 ﻿/**
  ^
Configuration.h:1:3: error: stray '\277' in program
 ﻿/**
   ^
In file included from J:\Marlin-2.0.x\Marlin\src\inc\MarlinConfigPre.h:39:0,
                 from J:\Marlin-2.0.x\Marlin\src\inc\MarlinConfig.h:28,
                 from J:\Marlin-2.0.x\Marlin\src\HAL\AVR\HAL.cpp:24:
Configuration.h:1:1: error: stray '\357' in program
 ﻿/**
 ^
Configuration.h:1:2: error: stray '\273' in program
 ﻿/**
  ^
Configuration.h:1:3: error: stray '\277' in program
 ﻿/**
   ^
In file included from J:\Marlin-2.0.x\Marlin\src\inc\MarlinConfigPre.h:39:0,
                 from J:\Marlin-2.0.x\Marlin\src\HAL\AVR\fast_pwm.cpp:24:
Configuration.h:1:1: error: stray '\357' in program
 ﻿/**
 ^
Configuration.h:1:2: error: stray '\273' in program
 ﻿/**
  ^
Configuration.h:1:3: error: stray '\277' in program
 ﻿/**
   ^
In file included from J:\Marlin-2.0.x\Marlin\src\inc\MarlinConfigPre.h:39:0,
                 from J:\Marlin-2.0.x\Marlin\src\inc\MarlinConfig.h:28,
                 from J:\Marlin-2.0.x\Marlin\src\HAL\AVR\HAL_SPI.cpp:34:
Configuration.h:1:1: error: stray '\357' in program
 ﻿/**
 ^
In file included from J:\Marlin-2.0.x\Marlin\src\inc\MarlinConfigPre.h:39:0,
                 from J:\Marlin-2.0.x\Marlin\src\inc\MarlinConfig.h:28,
                 from J:\Marlin-2.0.x\Marlin\src\HAL\AVR\eeprom.cpp:24:
Configuration.h:1:1: error: stray '\357' in program
 ﻿/**
 ^
Configuration.h:1:2: error: stray '\273' in program
 ﻿/**
  ^
Configuration.h:1:3: error: stray '\277' in program
 ﻿/**
   ^
Configuration.h:1:2: error: stray '\273' in program
 ﻿/**
  ^
Configuration.h:1:3: error: stray '\277' in program
 ﻿/**
   ^
In file included from J:\Marlin-2.0.x\Marlin\src\inc\MarlinConfigPre.h:39:0,
                 from J:\Marlin-2.0.x\Marlin\src\inc\MarlinConfig.h:28,
                 from J:\Marlin-2.0.x\Marlin\src\HAL\AVR\Servo.cpp:55:
Configuration.h:1:1: error: stray '\357' in program
 ﻿/**
 ^
Configuration.h:1:2: error: stray '\273' in program
 ﻿/**
  ^
Configuration.h:1:3: error: stray '\277' in program
 ﻿/**
   ^
"C:\\Users\\braun\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\avr-gcc\\7.3.0-atmel3.6.1-arduino7/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -MMD -flto -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10815 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR "-IC:\\Users\\braun\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\avr\\1.8.3\\cores\\arduino" "-IC:\\Users\\braun\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\avr\\1.8.3\\variants\\mega" "-IC:\\Program Files (x86)\\Arduino\\libraries\\LiquidCrystal\\src" "C:\\Users\\braun\\AppData\\Local\\Temp\\arduino_build_819598\\sketch\\src\\HAL\\AVR\\u8g_com_HAL_AVR_sw_spi.cpp" -o "C:\\Users\\braun\\AppData\\Local\\Temp\\arduino_build_819598\\sketch\\src\\HAL\\AVR\\u8g_com_HAL_AVR_sw_spi.cpp.o"
In file included from J:\Marlin-2.0.x\Marlin\src\inc\MarlinConfigPre.h:39:0,
                 from J:\Marlin-2.0.x\Marlin\src\HAL\AVR\u8g_com_HAL_AVR_sw_spi.cpp:58:
Configuration.h:1:1: error: stray '\357' in program
 ﻿/**
 ^
Configuration.h:1:2: error: stray '\273' in program
 ﻿/**
  ^
Configuration.h:1:3: error: stray '\277' in program
 ﻿/**
   ^
In file included from J:\Marlin-2.0.x\Marlin\src\inc\MarlinConfigPre.h:39:0,
                 from J:\Marlin-2.0.x\Marlin\src\inc\MarlinConfig.h:28,
                 from J:\Marlin-2.0.x\Marlin\src\HAL\AVR\watchdog.cpp:24:
Configuration.h:1:1: error: stray '\357' in program
 ﻿/**
 ^
Configuration.h:1:2: error: stray '\273' in program
 ﻿/**
  ^
Configuration.h:1:3: error: stray '\277' in program
 ﻿/**
   ^
Bibliothek LiquidCrystal in Version 1.0.7 im Ordner: C:\Program Files (x86)\Arduino\libraries\LiquidCrystal  wird verwendet
exit status 1
stray '\357' in program

### Version of Marlin Firmware

02000801

arnold-b, 2021-05-29:
Found my Problem. 
It only happen if i use the "Marlin Editor" for changing the config file. 
If i do it manualy, i dont get any error.

ellensp, 2021-05-30:
What is a "Marlin editor"?  I've never heard of it.

arnold-b, 2021-05-30:
MarlinEditor is coming with the Marlin3DprinterTool

I thought i solved it. 
Now i changed the last setting inside Arduino IDE  and compiled again 

#define X_HOME_DIR 1

and the compiling Problem comes again. 

Finaly it looks like i only can use notepad to change the config

ellensp, 2021-05-30:
Marlin3DprinterTool is not offical part of marlin at all. 
We cannot hep with this,

ellensp, 2021-05-30:
I sudject you create an issue here https://github.com/cabbagecreek/Marlin3DprinterTool/issues

closed 2021-05-30 by ellensp

arnold-b, 2021-05-30:
but the last setting i changed directly in arduino IDE and the problem happend again. 
#define X_HOME_DIR 1
Finaly it looks like it happen random