MarlinFirmware/Marlin #21998: ["BUG"] Auto level bed(BLTouch) does not compensate height enought

SmokeXXL, 2021-05-29:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

Hello, I am trying autolevel bed but it does not compensate enought. I have two corners about 1mm lower than others. It does not reach a right height.

### Bug Timeline

I find this problem today

### Expected behavior

Compensate height with measured data.

### Actual behavior

It compensate something, but it is not enought. I tried mesh bed leveling and also G29 in start Gcode. Same results

### Version of Marlin Firmware

2.0.8.1

### Printer model

Custom printer(cartesian)

### Electronics

SKR 1.4 Turbo

### Add-ons

BLTouch

### Your Slicer

Cura

### Additional information & file uploads

[config.zip](https://github.com/MarlinFirmware/Marlin/files/6565123/config.zip)

ontheronix, 2021-05-30:
Can you confirm you loaded defaults after flashing? M502
Can you confirm bed leveling is active? M420 S1
Furthermore, enable debug in firmware, read with M111 S32 and put contents here.

What is the size of your printbed? Some issues seem to occur with big beds ( https://github.com/MarlinFirmware/Marlin/issues/19132 )

SmokeXXL, 2021-05-30:
Yes, I loaded EEPROM every time after new flash. 
 
 Leveling bed is turn on and my
### start Gcode is:

> G21
> G90
> M82
> M104 S160
> M140 S{material_bed_temperature_layer_0}
> G28
> G0 X-20 Y25 F4000
> M190 S{material_bed_temperature_layer_0}
> M104 S{material_print_temperature_layer_0}
> M190 S{material_bed_temperature_layer_0}
> M109 S{material_print_temperature_layer_0}
> G0 Z20 F5000
> G0 X-20 Y3 F4000
> G92 E0
> G1 F200 E10
> G4 P10000
> G0 Z0.15
> G92 E0
> G1 X40 E25 F500
> G92 E0
> G1 E-1 F500
> G1 X80 F4000
> G21
> G90
> M82
> M420 S1
> M420 V1

Now I am at work, I will post debug data the day after tomorrow.

I just wanna say: Printer compensate some height. but it is not enought, because my bed is about 2mm uneven (working on this - need new +-flat base bed)

It compensate about 0.5mm correctly I believe, but I need more for now.

ontheronix, 2021-05-30:
Ok.
What is the size of your printbed?

SmokeXXL, 2021-05-30:
220x240

arielito-ear, 2021-06-09:
Hi, i have similar problem, mi printer is a Delta, 280mm diameter, i run G33 P1, P2 P3, P4 P7 P10 and algo G29 in different order, and always the left side is OK, but de right side touch the nozzle. 
Version 2.0.8.2.

ellensp, 2021-06-09:
Whenever there are homing or leveling issues, we now ask everyone to follow a standard procedure to gather more information so that we're not just taking stabs in the dark. Here is the boilerplate:

- Download Marlin `bugfix-2.0.x` to test with the latest code.
- Enable `DEBUG_LEVELING_FEATURE` and `M114_DETAIL` and re-flash the firmware.
- Connect to your printer from host software such as Cura, Printrun or Repetier Host.
- Send `M502` and `M500` to ensure your Configurations are applied.
- Issue the command `M111 S247` to enable maximum logging.
- Perform a `G28` to do your standard homing procedure.
- Do a `G29` to probe the bed. This will also enable bed leveling.
- Do some of the moves that revealed problems before. Take notes.
- Copy the log output into a **.TXT file** and attach it to your next reply.

Repeat this procedure, if needed, to demonstrate inconsistencies. From these logs we should hopefully get a better idea of what's going on with your machine.

arielito-ear, 2021-06-10:
Hi, G29, without parameters?

I ran G29 P0, G29 P1, G29 A, i must run this again or only G29 without parameters?

Thanks.

closed 2021-08-20 by github-actions[bot]