MarlinFirmware/Marlin #22006: Chamber heater control wont unlink to extruder hysteresis

OutcastZeroOne, 2021-05-30:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

SKR 1.4 Turbo
have tried multiple bugfix release of this week. 

Trying to configure settings for a chamber heater getting signal off the extra extruder output of the SKR. the issue is that no matter what i try to configure for chamber control it continues to link to the extruder setting and times out after a few seconds ignoring the 3min time limit I set.

Version of bugfix from April 17 worked fine with my old SKR 1.3. I tried to recompile that version initially but several libraries auto updated and then I continued to get the error about missmatched versions of config/config_adv.

### Expected behavior

turn on chamber heater, heater begins to heat chamber with a several minute timer for the full degree increase of temperature till target temperature is reached.

### Actual behavior

turn on chamber heater, heater begins to heat chamber, get thermal runaway error after a few seconds of heating, even if I disable thermal runaway protection for the chamber.

### Steps to Reproduce

try to heat chamber
runs for few seconds
throws Thermal runaway error, even if disabling thermal runaway setting

### Version of Marlin Firmware

Bugfix 2.0 released May 27, several others same week

### Printer model

Custom

### Electronics

SKR 1.4 Turbo, FYSETC_MINI_12864_2_1

### Add-ons

 custom chamber heater

### Additional information & file uploads

[ozomarlin.zip](https://github.com/MarlinFirmware/Marlin/files/6567032/ozomarlin.zip)

ellensp, 2021-05-31:
whay d you haev #define MAX_CHAMBER_POWER 208 ?
This trottles the heater, and is a likly cause of THERMAL_PROTECTION_CHAMBER wich is enabled in provided config files

OutcastZeroOne, 2021-05-31:
ive tried multiple things. bottom line the underlying issue is that its not recognizing the use of the second extruder output as chamber heater but still linking it to the other extruders settings. the fact that even when i disabled thermal runaway protection on the chamber and it still tripped it shows something isnt right.