MarlinFirmware/Marlin #22007: [BUG] (bug summary) Power off continuation function is not available in m605 image mode

cappuccino02, 2021-05-31:
### Did you test the latest `bugfix-2.0.x` code?

No, but I will test it now!

### Bug Description

DUAL_ X_ Cariage.Power off continuation function is not available in m605 image mode

### Bug Timeline

Dunno. Probably a new bug.

### Expected behavior

Continue printing unfinished files

### Actual behavior

Print from the beginning, not continue to print unfinished files

### Version of Marlin Firmware

2.0.8.1

### Electronics

BIGTREETECH SKR PRO 1.2

### Your Slicer

Cura

ellensp, 2021-06-01:
Please attach your configuration files.

cappuccino02, 2021-06-02:
抄送: "WEJIR   ENCJI   YA     ***@***.******@***.***&gt;;

cappuccino02, 2021-06-02:
> ------------------ 原始邮件 ------------------- 发件人: "MarlinFirmware/Marlin " ***@***.***>; 发送时间： 2021年6月1日（星期二）下午5:57 ***@***.***>; 抄送: "WEJIR ENCJI YA ***@***.******@***.***>;主题: Re: [MarlinFirmware/Marlin] [BUG] (bug摘要) m605 图像模式下无法使用断电续传功能 ( #22007 ) 请附上您的配置文件。-您收到此消息是因为您创作了该线程。直接回复此电子邮件，在 GitHub 上查看，或取消订阅。

cappuccino02, 2021-06-02:
[Uploading Marlin.ZIP…]()

ellensp, 2021-06-02:
> Uploading Marlin.ZIP…

This just links to https://github.com/MarlinFirmware/Marlin/issues/22007
no actual config file.

cappuccino02, 2021-06-02:
> > 正在上传 Marlin.ZIP...
> 
> 这只是链接到#22007
> 没有实际的配置文件。

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/6581682/Marlin.zip)

closed 2021-08-14 by github-actions[bot]