MarlinFirmware/Marlin #22009: installed marlin 2.0.8.2 for cr10 printer with cr10 config issue

mr47xx, 2021-05-31:
### Did you test the latest `bugfix-2.0.x` code?

No, but I will test it now!

### Bug Description

so basically i just got a skr mini e3 v2 that works with cr10 but when i flashed in with the marlin 2.0.8.2 with cr10 config when i auto home it it send the x all the way to the stopper the y all the way to the stopper and because i have bl touch if it were to be able to fix the x and y issue i assume it would try to push my nozzle through my glass bed corner  and i have no clue what to do or fix 

### Expected behavior

 to auto home at the left front corner  section 1 area were its suppose to 

### Actual behavior

 ramming the x into the x stopper and the bed into the y stopper

### Steps to Reproduce

turn on press auto home 

### Version of Marlin Firmware

2.0.8.0

### Printer model

cr10 

### Electronics

bigtreetech skr mini e3 v2

### Add-ons

only add on are bl touch and x and y smoothers

### Your Slicer

Cura

### Host Software

SD Card (headless)

ellensp, 2021-06-01:
This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://discuss.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2021-06-01 by ellensp

mr47xx, 2021-06-01:
thank aparently it had to do with these jumper clips attach to the board only issue now is the z moves up when auto homed instead of down but works moving up when manually moved