MarlinFirmware/Marlin #22014: [BUG] (Failed to Compile)

CorruptName, 2021-06-01:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

Cannot Compile with:
`#define MECHANICAL_GANTRY_CALIBRATION`
On BTT SKR 1.4 Turbo
I assume this was the problem with issue: #17545

### Expected behavior

Compile

### Actual behavior

Compile Failed

### Steps to Reproduce

1: Enable "MECHANICAL_GANTRY_CALIBRATION" in Configuration_adv
2: Compile

### Version of Marlin Firmware

Bugfix 2.0

### Electronics

BTT SKR 1.4 Turbo

ellensp, 2021-06-01:
No Configs, no error message
Difficult to work out what your talking about
Please attach your configuration files.

ellensp, 2021-06-01:
Setting   #define MOTHERBOARD BOARD_BTT_SKR_V1_4_TURBO
and #define MECHANICAL_GANTRY_CALIBRATION
Results in a deliberate error "It is highly recommended to have adjustable current drivers to prevent damage. Disable this line to continue anyway."
Where you have to do what it says.
I changed stepper driver types to TMC2209 (ie adjustable current drivers) and it comiles without issues

CorruptName, 2021-06-01:
> No Configs, no error message
> Difficult to work out what your talking about
> Please attach your configuration files.

The Error:
Linking .pio\build\LPC1769\firmware.elf
c:/users/corrupt/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld.exe: .pio/build/LPC1769/src/src/gcode/calibrate/G34_M422.cpp.o: in function `GCodeParser::has_value()':
C:\Users\Corrupt\Documents\Marlin-bugfix-2.0.x/Marlin\src\gcode\calibrate/G34_M422.cpp:70: multiple definition of `GcodeSuite::G34()'; .pio/build/LPC1769/src/src/gcode/calibrate/G34.cpp.o:C:\Users\Corrupt\Documents\Marlin-bugfix-2.0.x/Marlin\src\gcode\calibrate\../parser.h:110: first defined here
collect2.exe: error: ld returned 1 exit status
*** [.pio\build\LPC1769\firmware.elf] Error 1

Also. i already have all TMC2209.

CorruptName, 2021-06-01:
[Configuration.h.txt](https://github.com/MarlinFirmware/Marlin/files/6579830/Configuration.h.txt)
[Configuration_adv.h.txt](https://github.com/MarlinFirmware/Marlin/files/6579833/Configuration_adv.h.txt)

CorruptName, 2021-06-01:
I should also mention that I don't much care to solve this. If you're not worried, I'm not either.

thinkyhead, 2021-06-02:
Well, remember you're asking "someone in the world" to fix this, which is to say whoever happens upon this posting. I'll try your configuration files and see whether they compile okay with the current bugfix, and get back to you in a second….

thinkyhead, 2021-06-02:
The configurations you included are not the ones you used in testing the current `bugfix-2.0.x`. Do you have those handy? In the meantime, I will try to patch yours up for you so they are compatible with the current `bugfix-2.0.x`.

thinkyhead, 2021-06-02:
Indeed, zero extruders CNC configuration with Trinamic drivers was not tested, so I have to bring in some of the patches from #19112 into bugfix first, and then I might just be able to provide you with a working config.

thinkyhead, 2021-06-02:
Ok, so you'll have to disable either `MECHANICAL_GANTRY_CALIBRATION` or `Z_MULTI_ENDSTOPS`. This is because both use the same G-code, `G34`, but one is for mechanical endstops while the other is for sensorless homing. Otherwise, your patched up configuration is probably fine. But if you want to compare, here is what I ended up with…

[issue22014configs.tar.gz](https://github.com/MarlinFirmware/Marlin/files/6581519/issue22014configs.tar.gz)

closed 2021-06-02 by thinkyhead

CorruptName, 2021-06-02:
> The configurations you included are not the ones you used in testing the current `bugfix-2.0.x`. Do you have those handy? In >the meantime, I will try to patch yours up for you so they are compatible with the current `bugfix-2.0.x`.

Sorry, Didn't realize I wasn't on the latest bugfix. Built this machine of the last couple months and it must have been longer than I realized. Thanks for having a look.

>Well, remember you're asking "someone in the world" to fix this, which is to say whoever happens upon this posting. I'll try >your configuration files and see whether they compile okay with the current bugfix, and get back to you in a second….
I don't believe I have asked anything. Only brought a bug to your attention. As stated, I personally don't care if it gets fixed or not. I'm more interested in someone who cannot compile using this post to figure out why since the error wasn't exactly clear.

> Ok, so you'll have to disable either `MECHANICAL_GANTRY_CALIBRATION` or `Z_MULTI_ENDSTOPS`. This is because both use the same G-code, `G34`, but one is for mechanical endstops while the other is for [...]
Thanks again for having a look. I will definitely compare and maybe give it a try.