MarlinFirmware/Marlin #22015: [BUG] Y_BED_SIZE not observed

tmartin007, 2021-06-01:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

In configuration.h, size of the printable area can be specified, i.e. X_BED_SIZE, Y_BED_SIZE.

While G0 X or G1 X movements stop, once X >= X_BED_SIZE, the specified value for Y_BED_SIZE does not appear to be considered.

### Bug Timeline

something similar was observed when reporting issue #16568

### Expected behavior

No movements that would yield a y-position > Y_BED_SIZE should be executed.

### Actual behavior

When defining Y_BED_SIZE as 590, a G0 Y600 moves the Y-axis beyond 590. If the physical bed-length is actually less than 600 (e.g. 590), the Y axis bumps into the frame of the "printer" (in my case a 6040 CNC-mill).

### Steps to Reproduce

1) define Y_BED_SIZE to be 590 in configuration.h
2) build and install the firmware
3) execute G28 Y0 to home the Y-axis
4) execute G0 Y600
5) observe the Y axis bumping into the frame of the printer

### Version of Marlin Firmware

2.0.8.2

### Printer model

CNC 6040 mill

### Electronics

BTT SKR Pro V1.1, TMC 2209

### Additional information & file uploads

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/6575309/Configuration.zip)

DerAndere1, 2021-06-06:
did you send M502 followed by M500 after updating the firmware?

tmartin007, 2021-06-06:
no, I had not tried this . thanks for the hint, I will report the result shortly

tmartin007, 2021-06-24:
Thanks for the idea!

Unfortunately, this had no effect.

M502
echo:Hardcoded Default Settings Loaded
ok
M500
Error:EEPROM disabled
ok

closed 2021-09-04 by github-actions[bot]