MarlinFirmware/Marlin #22016: [BUG] Initial Homing does not work on X-axis and Z-axis

tmartin007, 2021-06-01:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

Issuing G28X or G28Z initially moves the respective axis a short distance into the direction that is opposite to the defined home-position and stops.

In order for G28X to actually perform homing (after a power up), it alway has to be invoked exactly 3 times. - This is reliably reproducible.

In order for G28Z to actually perform homing (after a power up), it alway has to be invoked exactly 2 times. - This is reliably reproducible.

In order for G28Y to actually perform homing (after a power up), it alway has to be invoked exactly once. - This is reliably reproducible - and is expected and desired.

### Bug Timeline

A similar behavior was reported with issue #17128

### Expected behavior

1) In order for G28X to actually perform homing (after a power up), a single invocation of G28X should be sufficient.

2) In order for G28Z to actually perform homing (after a power up), a single invocation of G28Z should be sufficient.

### Actual behavior

Issuing G28X or G28Z initially moves the respective axis a short distance into the direction that is opposite to the defined home-position and stops.

### Steps to Reproduce

1) boot-up Marlin 2.0.8.2 on a BTT SKR Pro V1.1 with TMC2209 on all axis
2) issue G28X to home the x-axis
3) observe the x-axis to move away from the home-position for a small distance (maybe: HOMING_BACKOFF_POST_MM) 
4) issue another G28X to home the x-axis
5) again observe the x-axis to move away from the home-position for a small distance (maybe: HOMING_BACKOFF_POST_MM) 
6) issue yet another G28X to home the x-axis
7) observe the x-axis perform the homing procedure with HOMING_BACKOFF_POST_MM, when done

8) issue G28Z to home the z-axis
9) observe the z-axis to move away from the home-position for a small distance (maybe: HOMING_BACKOFF_POST_MM) 
10) issue another G28Z to home the z-axis
11) observe the z-axis perform the homing procedure with HOMING_BACKOFF_POST_MM, when done

### Version of Marlin Firmware

2.0.8.2

### Printer model

CNC 6040 mill

### Electronics

BTT SKR Pro V1.1, TMC 2209

### Add-ons

TMC 2209 on all axis

### Additional information & file uploads

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/6575446/Configuration.zip)

ellensp, 2021-06-01:
What does M119 say?
as moving small disatnce in wrong direction is symtomatic of ndstop alsready triggered. 
It trys to home, cant as it already triggered. backs off, attrmts to rehome and cant as already triggered. net result moves away from home small distance.

tmartin007, 2021-06-01:
Thanks for you quick response.
After power-up, M119 says:

`M119
Reporting endstop status
x_min: open
y_min: open
z_min: open
ok`

We have tried M914 I Z20 to reduce the TMC Bump Sensitivity without achieving any improvement. (However, a sensitivity of 20 is way too low, to even trigger a "STALL"). - BTW, "HOME_BUMP_MM" is set to "0" for all axis.

ellensp, 2021-06-01:
May be related to https://github.com/MarlinFirmware/Marlin/issues/22012

tmartin007, 2021-06-01:
According to #22012, homing Z works, which unfortunately, I cannot confirm.

DerAndere1, 2021-06-11:
Have you sent M502 followed by M500 after uploading Marlin?

tmartin007, 2021-06-11:
No, the machine is currently disassembled, I am intending to put it together again,
tomorrow and/or on Sunday and will test this immediately.

Thanks again!

tmartin007, 2021-06-24:
Here too, thanks for the idea!

Unfortunately, this had no effect.

M502
echo:Hardcoded Default Settings Loaded
ok
M500
Error:EEPROM disabled
ok

closed 2021-09-04 by github-actions[bot]