MarlinFirmware/Marlin #22018: FR -INDEX mode improvements

DrumClock, 2021-06-02:
> Hi @InsanityAutomation
> I thought of adding to **Configuration_adv.h** settings for MIRRORED mode:
> 
> ```
> // Default x offset in mirrrored mode (typically set to print bed width)
> #define DEFAULT_MIRRORED_X_OFFSET X_BED_SIZE
> 
> // Set the maximum so that the first X-carriage cannot hit the second X-carriage
> #define X_MAX_MIRRORED_POS  95 
> ```
> 
> [image]
> 
> So when selecting **MIRRORED** mode from LCD or M605 S3, carriages is set to **X1 = 0**
> and **X2 = X_BED_SIZE** . `X_MAX_MIRRORED_POS` to **restrict movement** to avoid collision of carriages

InsanityAutomation, 2021-06-12:
Im making my way closer to this on my list..

DrumClock, 2021-07-14:
@InsanityAutomation 
Because something doesn't work with each new release of bugfix 2.0.x 
(what used to be OK) and nobody solves it, **I decided to switch to FW Klipper.**
**I will no longer pay attention to these issues.**

johncarlson21, 2023-02-14:
> Another thing to improvement on mirror mode is to adjust the center position. When switching to mirror mode and the printer moves everything into position my x1 is still off the bed while the x2 is 65mm on. Losing 65 on the right side of my 310 bed.

which version are you on? I just had this problem for my 2.1.x build.. I had to fix this in the src/module/motion.cpp
I have an X2_MAX_POS

from
new_pos.x = _MIN(X_BED_SIZE - x0_pos, X_MAX_POS);
to
new_pos.x = _MIN(X_BED_SIZE - x0_pos, X2_MAX_POS);