MarlinFirmware/Marlin #22024: [BUG] DETECT_BROKEN_ENDSTOP and Z-Probe as Z-Endstop (BLTouch)

fiechr, 2021-06-02:
### Did you test the latest `bugfix-2.0.x` code?

No, but I will test it now!

### Bug Description

I'm using a BL-Touch on three different printers. Until now I've used Marlin up to 2.0.8 and had enabled `DETECT_BROKEN_ENDSTOP` for all of them without any compile time errors. All printers don't have their Z-endstop-switches removed and the Z-probe (all BLTouch) is configured to act as end-stop, which worked without issues.

After upgrading to 2.0.8.2, I get this error when compiling:
```
Marlin/src/module/motion.cpp: In function 'void homeaxis(AxisEnum)':
Marlin/src/module/motion.cpp:1634:29: error: 'Z_ENDSTOP' was not declared in this scope
           case Z_AXIS: es = Z_ENDSTOP; break;
                             ^
At global scope:
cc1plus: warning: unrecognized command line option '-Wno-register'
*** [.pio/build/mega2560/src/src/module/motion.cpp.o] Error 1
```

If I comment out `DETECT_BROKEN_ENDSTOP` the error goes away.

Is this expected or a bug? Or in other words: Is this option usable with a Z probe installed which acts also as Z endstop?

### Bug Timeline

2.0.8.2

### Expected behavior

`DETECT_BROKEN_ENDSTOP` enabled compiles without errors when z probe is used as z endstop.

### Actual behavior

Compile time error as long as `DETECT_BROKEN_ENDSTOP` stays enabled.

### Steps to Reproduce

Printer configuration that uses z probe as z endstop.
Enable `DETECT_BROKEN_ENDSTOP`
Try to compile.

### Version of Marlin Firmware

2.0.8.2

### Printer model

Artillery Sidewinder X1, Anycubic i3 Mega S, Creality Ender 3 Pro

### Electronics

BLTouch (3DTouch) installed

### Your Slicer

Prusa Slicer

### Host Software

OctoPrint

DerAndere1, 2021-06-08:
@flechr : Can you test if the change described by @FanDjango in the comment at https://github.com/MarlinFirmware/Marlin/commit/4dae5890e99c73686b7e1ee08857487a0acfeb28#commitcomment-51183525 fixes this ? 
 
This 
https://github.com/MarlinFirmware/Marlin/blob/d320545066676338d4a0142268931bb66eedfccc/Marlin/src/module/endstops.h#L84

#if HAS_Z_MIN || HAS_Z_MAX

broke it again for my config. I have no Z_MIN, and I have no Z_MAX, I am homing with the probe. So I get a compile error, no Z_ENDSTOP defined.

I needed to change this to

#if HAS_Z_MIN || HAS_Z_MAX || HOMING_WITH_PROBE

fiechr, 2021-06-11:
No, the error persists when including `HOMING_WITH_PROBE` (still using the 2.0.x branch).

But if I use  `HOMING_Z_WITH_PROBE` it works. Not sure if that makes sense.

```
  // Endstops can be either MIN or MAX but not both
  #if HAS_X_MIN || HAS_X_MAX
    , X_ENDSTOP = TERN(X_HOME_TO_MAX, X_MAX, X_MIN)
  #endif
  #if HAS_Y_MIN || HAS_Y_MAX
    , Y_ENDSTOP = TERN(Y_HOME_TO_MAX, Y_MAX, Y_MIN)
  #endif
  #if HAS_Z_MIN || HAS_Z_MAX || HOMING_Z_WITH_PROBE
    , Z_ENDSTOP = TERN(Z_HOME_TO_MAX, Z_MAX, TERN(HOMING_Z_WITH_PROBE, Z_MIN_PROBE, Z_MIN))
  #endif
```

thinkyhead, 2021-06-11:
`HOMING_Z_WITH_PROBE` is correct. Patch applied.

closed 2021-06-11 by thinkyhead