MarlinFirmware/Marlin #22026: [BUG] HAS_MARLINUI_U8GLIB  = U8glib-HAL@~0.4.5

DrumClock, 2021-06-03:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

Po dnešní aktualizaci nelze kompilovat:

```
Library Manager: Installing U8glib-HAL @ ~0.4.5
KeyError: 'framework-stm32cube':
  File "C:\Users\Petr\.platformio\penv\Lib\site-packages\platformio\builder\main.py", line 177:
    env.SConscript("$BUILD_SCRIPT")
  File "C:\Users\Petr\.platformio\packages\tool-scons\scons-local-4.1.0\SCons\Script\SConscript.py", line 591:
    return _SConscript(self.fs, *files, **subst_kw)
  File "C:\Users\Petr\.platformio\packages\tool-scons\scons-local-4.1.0\SCons\Script\SConscript.py", line 280:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "C:\Users\Petr\.platformio\platforms\ststm32\builder\main.py", line 120:
    target_elf = env.BuildProgram()
  File "C:\Users\Petr\.platformio\packages\tool-scons\scons-local-4.1.0\SCons\Util.py", line 658:
    return self.method(*nargs, **kwargs)
  File "C:\Users\Petr\.platformio\penv\Lib\site-packages\platformio\builder\tools\platformio.py", line 62:
    env.ProcessProjectDeps()
  File "C:\Users\Petr\.platformio\packages\tool-scons\scons-local-4.1.0\SCons\Util.py", line 658:
    return self.method(*nargs, **kwargs)
  File "C:\Users\Petr\.platformio\penv\Lib\site-packages\platformio\builder\tools\platformio.py", line 140:
    project_lib_builder = env.ConfigureProjectLibBuilder()
  File "C:\Users\Petr\.platformio\packages\tool-scons\scons-local-4.1.0\SCons\Util.py", line 658:
    return self.method(*nargs, **kwargs)
  File "C:\Users\Petr\.platformio\penv\Lib\site-packages\platformio\builder\tools\piolib.py", line 1062:
    project.install_dependencies()
  File "C:\Users\Petr\.platformio\penv\Lib\site-packages\platformio\builder\tools\piolib.py", line 898:
    lm.install(spec)
  File "c:\users\petr\.platformio\penv\lib\site-packages\platformio\package\manager\_install.py", line 48:
    pkg = self._install(
  File "c:\users\petr\.platformio\penv\lib\site-packages\platformio\package\manager\library.py", line 93:
    if is_builtin_lib(spec.name):
  File "c:\users\petr\.platformio\penv\lib\site-packages\platformio\commands\lib\helpers.py", line 49:
    for storage in storages or get_builtin_libs():
  File "c:\users\petr\.platformio\penv\lib\site-packages\platformio\commands\lib\helpers.py", line 34:
    for storage in p.get_lib_storages():
  File "c:\users\petr\.platformio\penv\lib\site-packages\platformio\platform\base.py", line 214:
    pkg = self.get_package(opts["package"])
  File "c:\users\petr\.platformio\penv\lib\site-packages\platformio\platform\_packages.py", line 30:
    return self.pm.get_package(spec or self.get_package_spec(name))
  File "c:\users\petr\.platformio\penv\lib\site-packages\platformio\platform\_packages.py", line 22:
    owner=self.packages[name].get("owner"),
================================================================ [FAILED] Took 2.58 seconds ================================================================

Environment    Status    Duration
-------------  --------  ------------
rumba32        FAILED    00:00:02.575
=========================================================== 1 failed, 0 succeeded in 00:00:02.575 ===========================================================The terminal process "C:\Users\Petr\.platformio\penv\Scripts\platformio.exe 'run'" terminated with exit code: 1.

Pokud vrátím změnu na  = U8glib-HAL@~0.4.4 je to OK

```

### Version of Marlin Firmware

bugfix-2.0.x  (ze dne 3.6.2021)

ellensp, 2021-06-03:
Please attach your configuration files.

DrumClock, 2021-06-03:
[config files.zip](https://github.com/MarlinFirmware/Marlin/files/6589618/config.files.zip)

ellensp, 2021-06-03:
Provided configs and other changed files, compiles without issues for me but. not always... 

there is a warning which I think is part of the trigger
```
Library Manager: Warning! More than one package has been found by U8glib-HAL @ ~0.4.4 requirements:
 - marlinfirmware/U8glib-HAL @ 0.4.4
 - thinkyhead/U8glib-HAL @ 0.4.5 
```
This seems to sometimes trigger a bug in platformio

If you force it to use one or the other , it seem to compile more reliably
eg in features.ini change 
HAS_MARLINUI_U8GLIB                    = U8glib-HAL@~0.4.4
to 
HAS_MARLINUI_U8GLIB                    = marlinfirmware/U8glib-HAL@~0.4.4

ellensp, 2021-06-03:
@thinkyhead an you unplublish your thinkyhead/U8glib-HAL @ 0.4.5  from platformio?

DrumClock, 2021-06-03:
po další aktualizaci je to OK

```
Converting Marlin.ino
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Library Manager: U8glib-HAL @ 0.4.5 is already installed
Found 17 compatible libraries
Scanning dependencies...
Dependency Graph
```

closed 2021-06-03 by DrumClock