MarlinFirmware/Marlin #22027: [BUG] softserial broken on MKS robin nano 1.2

th0m4sek, 2021-06-03:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

```
In file included from Marlin/src/HAL/STM32/../../inc/MarlinConfigPre.h:37,
                 from Marlin/src/HAL/STM32/../../inc/MarlinConfig.h:28,
                 from Marlin/src/HAL/STM32/timers.cpp:24:
Marlin/src/HAL/STM32/timers.cpp: In function 'void SetTimerInterruptPriorities()':
Marlin/src/HAL/STM32/timers.cpp:190:44: error: 'setInterruptPriority' is not a member of 'SoftwareSerial'
  190 |   TERN_(HAS_TMC_SW_SERIAL, SoftwareSerial::setInterruptPriority(SWSERIAL_TIMER_IRQ_PRIO, 0));
      |                                            ^~~~~~~~~~~~~~~~~~~~
Marlin/src/HAL/STM32/../../inc/../core/macros.h:549:26: note: in definition of macro 'THIRD'
  549 | #define THIRD(a,b,c,...) c
      |                          ^
Marlin/src/HAL/STM32/../../inc/../core/macros.h:195:29: note: in expansion of macro '___TERN'
  195 | #define __TERN(T,V...)      ___TERN(_CAT(_NO,T),V)  // Prepend '_NO' to get '_NOT_0' or '_NOT_1'
      |                             ^~~~~~~
Marlin/src/HAL/STM32/../../inc/../core/macros.h:194:29: note: in expansion of macro '__TERN'
  194 | #define _TERN(E,V...)       __TERN(_CAT(T_,E),V)    // Prepend 'T_' to get 'T_0' or 'T_1'
      |                             ^~~~~~
Marlin/src/HAL/STM32/../../inc/../core/macros.h:193:29: note: in expansion of macro '_TERN'
  193 | #define TERN_(O,A)          _TERN(_ENA_1(O),,A)     // OPTION ? 'A' : '<nul>'
      |                             ^~~~~
Marlin/src/HAL/STM32/timers.cpp:190:3: note: in expansion of macro 'TERN_'
  190 |   TERN_(HAS_TMC_SW_SERIAL, SoftwareSerial::setInterruptPriority(SWSERIAL_TIMER_IRQ_PRIO, 0));
      |   ^~~~~
*** [.pio/build/mks_robin_nano35/src/src/HAL/STM32/timers.cpp.o] Error 1
```

### Bug Timeline

new

### Version of Marlin Firmware

bugfix-2.0.x

### Additional information & file uploads

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/6589899/Marlin.zip)

ellensp, 2021-06-03:
Cannot reproduce this error using provided config files and pins changes

th0m4sek, 2021-06-03:
forget to mention env: mks_robin_nano35
compilation on mks_robin_nano35_maple is working

ellensp, 2021-06-03:
I tried both. both compiled

th0m4sek, 2021-06-03:
i cloned repository and now i working fine sorry

closed 2021-06-03 by th0m4sek