MarlinFirmware/Marlin #22029: [FR] TMC drivers with communication: Check more registers to catch more errors

hamster65, 2021-06-03:
### Is your feature request related to a problem? Please describe.

One of my TMC2209 drivers changes from e.g. 128 microsteps to 8 during print. This causes stalls or the affected axis to repeadedly hit the mechanical limits until someone hears it an reacts. Obvously, the print goes wrong, material is wasted and mechanical wear or damage occurs.

This is a problem of the motor driver because it always affects the same TMC driver, regardless of it's position on the MGS SGEN L board.

### Are you looking for hardware support?

No, but I'd of course welcome any advice on how to fix the TMC board-

### Describe the feature you want

In addition to the existing TMC error checks (bad communication, overtemperature...) Marlin could check if other values change unexpectedly. For example the registers for microstepping.

### Additional context

Do you want to see my config and Output of M122 before and after the issue happens?

thisiskeithb, 2021-06-03:
If the error condition follows the driver, replace the driver. That’s not something Marlin can fix.

<hr>

This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- [Marlin Documentation](https://marlinfw.org) hosted on MarlinFW.org
- [Marlin Forum](https://forums.reprap.org/list.php?415) hosted on RepRap.org
- [Marlin Firmware](https://www.facebook.com/groups/1049718498464482/) Facebook Group 
- [Marlin Firmware for 3D Printers](https://www.facebook.com/groups/3Dtechtalk/) Facebook Group
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- [Marlin Firmware](https://discord.gg/n5NJ59y) Discord server

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2021-06-03 by thisiskeithb

hamster65, 2021-06-03:
Describe the feature you want

In addition to the existing TMC error checks (bad communication, overtemperature...) Marlin could check if other values change unexpectedly. For example the registers for microstepping.