MarlinFirmware/Marlin #22033: [BUG] marlin crashes on printing from SD card if power loss recovery enabled

maheshhchandra, 2021-06-03:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

marlin crashes on printing from SD card if power loss recovery enabled
using marlin 2.0.8.2
also checked on latest bugfix

Board 
SKR mini E3 V 1.2
marlin auto build :
stm32f103rc_btt_512k
stm32f103rc_btt_512k_stm32

was working with marlin 2.0.7.2

using BL touch
Filament runout sensor
on ender 3 pro
my conf files
[conf.zip](https://github.com/MarlinFirmware/Marlin/files/6593458/conf.zip)

### Steps to Reproduce

start print fron SD card if power loss recovery is switched on

### Version of Marlin Firmware

2.0.8.2

### Printer model

Ender 3 pro

### Electronics

SKR mini e3 v1.2

### Add-ons

BL touch, filament runout sensor

### Additional information & file uploads

[conf.zip](https://github.com/MarlinFirmware/Marlin/files/6593474/conf.zip)

ellensp, 2021-06-04:
Unless you main CPU on the controller is labeled as a STM32F103RE, using the stm32f103rc_btt_512k or stm32f103rc_btt_512k_stm32 is not guaranteed to be reliable at all.

Your are using flash memory from 256-512K that the chip manufacturer has not tested and is not guaranteed to work at all. 

Check your chip. If it says STM32f103RC, do no use 512k build environments and report bugs. These are most likely faults in your flash memory hardware.

When you have a STM32f103RC CPU the Firmware can only be accurately tested on Tested in build environments stm32f103rc_btt or stm32f103rc_btt_stm32   (now called  stm32f103rc_btt_maple and stm32f103rc_btt_

closed 2021-06-04 by ellensp

maheshhchandra, 2021-06-06:
error persists with build environment
stm32f103rc_btt 
stm32f103rc_btt_stm32 

attaching PLR file created also
[plr.zip](https://github.com/MarlinFirmware/Marlin/files/6604188/plr.zip)