MarlinFirmware/Marlin #22034: [BUG] Servo does extra rotation

Berkays, 2021-06-03:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

I'm using a custom optical z probe which has a servo to deploy the probe. Since Marlin 2.0.8 servo does some weird extra rotation even if set to same angle consecutively. Doesn't seem to be gcode issue as it happens with m280, m281 and at machine boot where it sets the deploy position. This issue doesn't occur in Marlin 2.0.7.2. I checked the commit history for servo related files but there isn't anything suspicious so maybe it's caused by changes in timer functions?

### Bug Timeline

Since Marlin 2.0.8

### Expected behavior

Servo should stay in same position if set with current angle position. It doesn't have to be the same angle, if servo is on position 100 and you set position 101 it currently does some weird 360 rotation

### Actual behavior

Does extra 360(?) degree motion even if position is same or very small

### Steps to Reproduce

Use marlin 2.0.8 or newer.

### Version of Marlin Firmware

2.0.8, bugfix

### Printer model

Creality Ender 3 Pro

### Electronics

SKR Mini E3 V1.2

### Add-ons

TowerPro SG90 Servo

### Additional information & file uploads

Configuration files:
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/6594048/Configuration.zip)

Video demonstrating issue. Note that the video is taken on second consecutive boot so servo position was already set to deploy angle.
https://user-images.githubusercontent.com/14983952/120704501-53e13d80-c4bf-11eb-9d17-44489b374e58.mp4

closed 2021-06-11 by Berkays

thisiskeithb, 2021-06-11:
> Solved by changing enviroment from `STM32F103RC_btt_512K` to `STM32F103RC_btt_512K_stm32`

Be sure to download the latest `bugfix-2.0.x` since `STM32F103RC_btt_512K` & `STM32F103RC_btt_512K_stm32` no longer exists. You'll use `STM32F103RC_btt` or `STM32F103RC_btt_usb` for non-maple builds.

Berkays, 2021-06-13:
Will do thanks.