MarlinFirmware/Marlin #22035: [BUG] Extendable MMUs with not precisely 5 slots fail with compile error from SanityCheck.h

codingcatgirl, 2021-06-03:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

When setting `#define MMU_MODEL EXTENDABLE_EMU_MMU2S` in the configuration, which refers to `MMUS with configurable number of filaments`, and set `EXTRUDERS` to a number that is not 5, i get the following compile error:

> PRUSA_MMU2(S) requires exactly 5 EXTRUDERS. Please update your Configuration

Additionally, if you set the number of extruders to more than 8 (which should be possible, since up to 15 are supported for an extendable MMU [accoring to the source code](https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.0.x/Marlin/src/inc/SanityCheck.h#L972), i additionally get the following compile error:

> Marlin supports a maximum of 8 EXTRUDERS.

Both of these messages shouldn't appear. I've checked the bugfix branch and the problems persist there.

### Bug Timeline

The bug of extendable MMUs actually not being extendable has, as far i can see, been present since the feature of extendable MMU2 was inroduced.

### Expected behavior

I should be able to run an extendable MMU with a different number than 5 slots, and even with more than 8 slots.

### Actual behavior

Marlin fails during debug time because the MMU doesn't have precisely 5 slots and because there are more slots than Marlin supposedly supports.

### Steps to Reproduce

1. Set `#define MMU_MODEL EXTENDABLE_EMU_MMU2S` in `Configuration.h`
2. Set `#define EXTRUDERS 10` in `Configuration.h`
3. Try to compile, and observe `SanityCheck.h` throwing errors.

### Version of Marlin Firmware

2.0.8.2

### Additional information & file uploads

They issue can be quickly traced down here:

In [this code](https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.0.x/Marlin/src/inc/Conditionals_LCD.h#L510), when `MMU_MODEL EXTENDABLE_EMU_MMU2S` is set, it results in the following values being defined:

```
#define HAS_PRUSA_MMU2 1    
#define HAS_PRUSA_MMU2S 1  
#define HAS_EXTENDABLE_MMU 1
```

However, neither [here](https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.0.x/Marlin/src/inc/SanityCheck.h#L959) nor [here](https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.0.x/Marlin/src/inc/SanityCheck.h#L981) is `HAS_EXTENDABLE_MMU` actually checked before throwing a compile error. 

I'm a bit surprised that i seem to be the first person noticing this and i'm wondering if i'm the first person to actually try to run a MMU with more than 5 slots on Marlin, since this problem seems to have already been there [in the commit that introduced this feature](https://github.com/MarlinFirmware/Marlin/commit/41529b65988a58ba512977e0fe1692aaeeb6c811) (#19912).

It would be nice if this issue could be fixed please :) Paging @GMagician because they wrote this specific feature. (Thank you for your work! I have designed my own extendable MMU and was so happy to see that i wouldn't have to make changes to Marlin because this feature was just recently added… well, almost :P)

codingcatgirl, 2021-06-03:
I just commented out these compile errors to see if it will just work then, and the answer is sadly no.

```
In file included from Marlin/src/inc/MarlinConfigPre.h:37:0,
                 from Marlin/src/inc/MarlinConfig.h:28,
                 from Marlin/src/MarlinCore.h:24,
                 from Marlin/src/MarlinCore.cpp:31:
Marlin/src/MarlinCore.cpp: In function 'void disable_e_stepper(uint8_t)':
Marlin/src/MarlinCore.cpp:318:52: error: 'DISABLE_AXIS_E8' was not declared in this scope
   #define _CASE_DIS_E(N) case N: DISABLE_AXIS_E##N(); break;
```

Marlin tries to use code/macors for extruders with index > 7 which unsurprisingly don't exist, since Marlin doesn't support more than 8 extruders. This happens despite `SINGLENOZZLE` being defined (but it gets defined [here](https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.0.x/Marlin/src/inc/Conditionals_LCD.h#L588) so that shouldn't make a difference). And yep, i've also uncommented the `#undef SINGLENOZZLE` from [this line](https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.0.x/Marlin/src/inc/SanityCheck.h#L958) – why is that line there, looks like a mistake to me?

It really seems like there might be quite a bit of work still to do in order to make this feature actually work? I can't help but get the impression that the extendability that the code/pull request promised to introduce wasn't tested at all.

GMagician, 2021-06-03:
>  I can't help but get the impression that the extendability that the code/pull request promised to introduce wasn't tested at all.

I stopped to 5 actually because my MMU was extendable once but just with standard colors

thisiskeithb, 2021-07-23:
Fixed in https://github.com/MarlinFirmware/Marlin/pull/22036

closed 2021-07-23 by thisiskeithb