MarlinFirmware/Marlin #22041: @madmax2 — Full instructions for ARM installation are now posted at http://marlinfw.org/docs/basics/install_arm.html

madmax2, 2021-06-04:
@madmax2 — Full instructions for ARM installation are now posted at http://marlinfw.org/docs/basics/install_arm.html

_Originally posted by @thinkyhead in https://github.com/MarlinFirmware/Marlin/issues/8618#issuecomment-382956085_

Arm instruction is missing from that link
it is just showing 404 page

thisiskeithb, 2021-06-04:
> Arm instruction is missing from that link

That comment was from 2018, so links have changed.

You can always visit the main [marlinfw.org](https://marlinfw.org) website or visit the Build and Install instructions directly: https://marlinfw.org/docs/basics/install.html#build-and-install.

closed 2021-06-04 by thisiskeithb

madmax2, 2021-06-05:
@thisiskeithb is this the new link?
https://marlinfw.org/docs/basics/install_rearm.html

BTW why you labelled this Bug? False Alarm.

I never said it was a bug..
Just trying to get the new updated link.

thisiskeithb, 2021-06-05:
> Just trying to get the new updated link and also to alert anyone reading of the previous thread
that the link is 404 now.

See my comment above.

<hr>

This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- [Marlin Documentation](https://marlinfw.org) hosted on MarlinFW.org
- [Marlin Forum](https://forums.reprap.org/list.php?415) hosted on RepRap.org
- [Marlin Firmware](https://www.facebook.com/groups/1049718498464482/) Facebook Group 
- [Marlin Firmware for 3D Printers](https://www.facebook.com/groups/3Dtechtalk/) Facebook Group
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- [Marlin Firmware](https://discord.gg/n5NJ59y) Discord server