MarlinFirmware/Marlin #22043: [BUG] SWITCHING_EXTRUDER - fail servo switching

DrumClock, 2021-06-04:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

při použití funkce :
```
// A dual extruder that uses a single stepper motor
#define SWITCHING_EXTRUDER
#if ENABLED(SWITCHING_EXTRUDER)
  #define WITH_NOZZLE  true    // one servo switches extruder and nozzles
  #define SWITCHING_EXTRUDER_SERVO_NR 1
  #define SWITCHING_EXTRUDER_SERVO_ANGLES { 45, 135, 45, 135 } // Angles for E0, E1[, E2, E3]
  #if EXTRUDERS > 3
    #define SWITCHING_EXTRUDER_E23_SERVO_NR 2
  #endif
#endif
```
Po odeslání příkazu  T0-T3  nepřepíná servo 1 a servo 2,  směr otáčení motoru E0/E1 je v pořádku.
Po odeslání příkazu M280  P1 nebo P2 obě serva pracují v pořádku.

Dále musí být funkce `#define DISABLE_INACTIVE_EXTRUDER` zakázána, protože
po krátkém čase přestane fungovat motor E1 pro T2 a T3.

### Version of Marlin Firmware

bugfix-2.0.x  (ze dne 4.6.2021)

### Printer model

special:  4x hotend +  2x motor extruderu

### Electronics

MKS Rumba32 

### Additional information & file uploads

[rumba32rew.zip](https://github.com/MarlinFirmware/Marlin/files/6596295/rumba32rew.zip)

DrumClock, 2021-06-07:
Ahoj,
ani dnešní update bugfix 2.0 (7.6.2021) neřeší chybu ovládání serva při použití funkce `SWITCHING_EXTRUDER`
Krokový motor funguje správně ale servo se při použití příkazu Tx vůbec nepohybuje.

Při použití M280 Px ......serva pracují dle očekávání.

thinkyhead, 2021-06-09:
Let's collect more data.

- Download Marlin `bugfix-2.0.x` to test with the latest code.
- Enable `DEBUG_TOOL_CHANGE` and re-flash the firmware.
- Connect to your printer from host software such as Cura, Printrun or Repetier Host.
- Send `M502` and `M500` to ensure your Configurations are applied.
- Do some of the commands that exhibited problems. Take notes.
- Copy the log output into a **.TXT file** and attach it to your next reply.

Repeat this procedure, if needed, to demonstrate inconsistencies. From these logs we should hopefully get a better idea of what's going on with your machine.

DrumClock, 2021-06-27:
Hi @thinkyhead  21 days ago I reported a bug to Github https://github.com/MarlinFirmware/Marlin/issues/22043
Today's update of bugfix2.0x did not fix anything.
Can someone please look at it and fix it?
The SWITCHIN_EXRUDER function is useless without a functional servo ....

DrumClock, 2021-07-14:
@thinkyhead 
Because something doesn't work with each new release of bugfix 2.0.x 
(what used to be OK) and nobody solves it, **I decided to switch to FW Klipper.**
**I will no longer pay attention to these issues.**

thinkyhead, 2021-07-30:
@DrumClock — Excellent! I spend all day every day on Marlin issues, and still there are always more, and I personally have no time to dive into most issues personally. So, thanks for being so understanding.

thinkyhead, 2021-07-30:
@InsanityAutomation — I have been unable to attend to this issue in spite of the demands of the entitled. This issue seems like it might be more your area, being that you are the tool-change wizard. What do you suppose is going on here with `SWITCHING_EXTRUDER`? Is this a genuine bug report or is it a hoax?

InsanityAutomation, 2021-07-30:
I don't have a 4 hotend version like this, but I can see what 2 does. I havnt updated the switching one in awhile.

thinkyhead, 2021-07-31:
Hmm, okay. Thanks for taking the trouble to respond in the common tongue. I am still not able to guess what is going on, but I will translate the text above and continue trying to find the source of the problem by staring at the code.

thinkyhead, 2021-07-31:
It looks like #22036 (June 8) included a patch for `DISABLE_INACTIVE_EXTRUDER` + `SWITCHING_EXTRUDER`. I don't have any hardware to confirm that the correct extruder(s) are being put to sleep / woken up by this change, or to check whether extruders are being put to sleep in error.

thinkyhead, 2021-07-31:
@InsanityAutomation — I get the feeling that our overbearing friend's report is indicating something about servos generally. Maybe that servos can go to sleep, and when they are told to move they are not powered so they do nothing. Or maybe they are remembering their angle, and not moving because they think they are already there.

Try sending `M280 P0 S-1` to power off the servo, then try T0 / T1 to switch the extruder. If it doesn't behave, that is a clue. The report hints that the servos are already powered off at startup, so you might not even need the `M280` command to see the issue.

closed 2022-04-12 by github-actions[bot]