MarlinFirmware/Marlin #22053: multi-axis compile issue

ellensp, 2021-06-05:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

I was just testing if I could add I axis to a ramps and got an error 

```
Compiling .pio/build/mega2560/src/src/module/tool_change.cpp.o
In file included from Marlin/src/module/../inc/MarlinConfigPre.h:37:0,
                 from Marlin/src/module/../inc/MarlinConfig.h:28,
                 from Marlin/src/module/../MarlinCore.h:24,
                 from Marlin/src/module/planner.h:33,
                 from Marlin/src/module/planner.cpp:65:
Marlin/src/module/planner.cpp: In static member function 'static void Planner::autotemp_task()':
Marlin/src/module/planner.cpp:1454:82: error: expected ')' before 'block'
       if (LINEAR_AXIS_GANG(block->steps.x, || block->steps.y, || block->steps.z, block->steps.i, || block->steps.j, || block->steps.k)) {

```

### Bug Timeline

new

### Expected behavior

I believe I have everything set correctly, it should compile

### Actual behavior

Marlin/src/module/planner.cpp: In static member function 'static void Planner::autotemp_task()':
Marlin/src/module/planner.cpp:1454:82: error: expected ')' before 'block'

### Steps to Reproduce

1. grab latest bugfix
2. On a ramp add a I axis  (use my attached configs)
3. build

### Version of Marlin Firmware

latest bugfix

### Additional information & file uploads

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/6602471/Configuration.zip)

I suspect all it needs is in Marlin/src/module/planner.cpp line 1454

replace 
`      if (LINEAR_AXIS_GANG(block->steps.x, || block->steps.y, || block->steps.z, block->steps.i, || block->steps.j, || block->steps.k)) {
`
with

`      if (LINEAR_AXIS_GANG(block->steps.x, || block->steps.y, || block->steps.z, || block->steps.i, || block->steps.j, || block->steps.k)) {
`

It compiles after this

ellensp, 2021-06-05:
@DerAndere1 first bug?

DerAndere1, 2021-06-05:
Thanks. I found some similar typos during development of 9 axes extension. Will open a small separate pull request soon.

thisiskeithb, 2021-06-05:
I'm also now getting errors in my sensorless homing configs:

```prolog
Compiling .pio/build/LPC1768/src/src/feature/bedlevel/abl/abl.cpp.o
In file included from Marlin/src/HAL/LPC1768/../../inc/MarlinConfig.h:49,
                 from Marlin/src/HAL/LPC1768/HAL.cpp:24:
Marlin/src/HAL/LPC1768/../../inc/SanityCheck.h:1702:19: error: redefinition of 'constexpr const float sbm []'
 1702 |   constexpr float sbm[] = SENSORLESS_BACKOFF_MM;
      |                   ^~~
Marlin/src/HAL/LPC1768/../../inc/SanityCheck.h:573:17: note: 'constexpr const float sbm [4]' previously defined here
  573 | constexpr float sbm[] = AXIS_RELATIVE_MODES;
      |                 ^~~
Marlin/src/HAL/LPC1768/../../inc/SanityCheck.h:1703:28: error: static assertion failed: SENSORLESS_BACKOFF_MM must have X Y Z elements (and no others).
 1703 |   static_assert(COUNT(sbm) == LINEAR_AXES, "SENSORLESS_BACKOFF_MM must have " _LINEAR_AXES_STR "elements (and no others).");
In file included from Marlin/src/HAL/LPC1768/../../inc/MarlinConfig.h:49,
                 from Marlin/src/HAL/LPC1768/HAL_MinSerial.cpp:24:
Marlin/src/HAL/LPC1768/../../inc/SanityCheck.h:1702:19: error: redefinition of 'constexpr const float sbm []'
 1702 |   constexpr float sbm[] = SENSORLESS_BACKOFF_MM;
      |                   ^~~
```

`SENSORLESS_BACKOFF_MM` only has two elements in the config & adding more doesn't work.

arminth, 2021-06-05:
#22055 didn't do it for me.

Fixed some but not all errors.

`Compiling .pio\build\LPC1769\src\src\HAL\LPC1768\HAL_SPI.cpp.o
Compiling .pio\build\LPC1769\src\src\HAL\LPC1768\MarlinSerial.cpp.o
In file included from Marlin\src\HAL\LPC1768\../../inc/MarlinConfig.h:49,
                 from Marlin\src\HAL\LPC1768\HAL_MinSerial.cpp:24:
Marlin\src\HAL\LPC1768\../../inc/SanityCheck.h:1702:28: error: static assertion failed: SENSORLESS_BACKOFF_MM must have X Y Z elements (and no others).
 1702 |   static_assert(COUNT(sbm) == LINEAR_AXES, "SENSORLESS_BACKOFF_MM must have " _LINEAR_AXES_STR "elements (and no others).");
In file included from Marlin\src\HAL\LPC1768\../../inc/MarlinConfigPre.h:37,
                 from Marlin\src\HAL\LPC1768\../../inc/MarlinConfig.h:28,
                 from Marlin\src\HAL\LPC1768\HAL_MinSerial.cpp:24:
Marlin\src\HAL\LPC1768\../../inc/SanityCheck.h:1706:31: error: non-constant condition for static assertion
 1706 |     static_assert(sbm[Z_AXIS] >= 0, "SENSORLESS_BACKOFF_MM.Z must be greater than or equal to 0."),
      |                   ~~~~~~~~~~~~^~~~
Marlin\src\HAL\LPC1768\../../inc/../core/macros.h:260:34: note: in definition of macro 'CODE_3'
  260 | #define CODE_3( A,B,C,...) A; B; C
      |                                  ^
Marlin\src\HAL\LPC1768\../../inc/../core/macros.h:264:24: note: in expansion of macro '_CODE_N'
  264 | #define CODE_N(N,V...) _CODE_N(N,V)
      |                        ^~~~~~~
Marlin\src\HAL\LPC1768\../../inc/../core/types.h:43:32: note: in expansion of macro 'CODE_N'
   43 | #define LINEAR_AXIS_CODE(V...) CODE_N(LINEAR_AXES, V)
      |                                ^~~~~~
Marlin\src\HAL\LPC1768\../../inc/SanityCheck.h:1703:3: note: in expansion of macro 'LINEAR_AXIS_CODE'
 1703 |   LINEAR_AXIS_CODE(
      |   ^~~~~~~~~~~~~~~~
Marlin\src\HAL\LPC1768\../../inc/SanityCheck.h:1706:31: error: array subscript value '2' is outside the bounds of array type 'const float [2]'
 1706 |     static_assert(sbm[Z_AXIS] >= 0, "SENSORLESS_BACKOFF_MM.Z must be greater than or equal to 0."),
      |                   ~~~~~~~~~~~~^~~~
Marlin\src\HAL\LPC1768\../../inc/../core/macros.h:260:34: note: in definition of macro 'CODE_3'
  260 | #define CODE_3( A,B,C,...) A; B; C
      |                                  ^
Marlin\src\HAL\LPC1768\../../inc/../core/macros.h:264:24: note: in expansion of macro '_CODE_N'
  264 | #define CODE_N(N,V...) _CODE_N(N,V)
      |                        ^~~~~~~
Marlin\src\HAL\LPC1768\../../inc/../core/types.h:43:32: note: in expansion of macro 'CODE_N'
   43 | #define LINEAR_AXIS_CODE(V...) CODE_N(LINEAR_AXES, V)
      |                                ^~~~~~
Marlin\src\HAL\LPC1768\../../inc/SanityCheck.h:1703:3: note: in expansion of macro 'LINEAR_AXIS_CODE'
 1703 |   LINEAR_AXIS_CODE(
      |   ^~~~~~~~~~~~~~~~
In file included from Marlin\src\HAL\LPC1768\../../inc/MarlinConfig.h:49,
                 from Marlin\src\HAL\LPC1768\HAL.cpp:24:
Marlin\src\HAL\LPC1768\../../inc/SanityCheck.h:1702:28: error: static assertion failed: SENSORLESS_BACKOFF_MM must have X Y Z elements (and no others).
 1702 |   static_assert(COUNT(sbm) == LINEAR_AXES, "SENSORLESS_BACKOFF_MM must have " _LINEAR_AXES_STR "elements (and no others).");
In file included from Marlin\src\HAL\LPC1768\../../inc/MarlinConfigPre.h:37,
                 from Marlin\src\HAL\LPC1768\../../inc/MarlinConfig.h:28,
                 from Marlin\src\HAL\LPC1768\HAL.cpp:24:
Marlin\src\HAL\LPC1768\../../inc/SanityCheck.h:1706:31: error: non-constant condition for static assertion
 1706 |     static_assert(sbm[Z_AXIS] >= 0, "SENSORLESS_BACKOFF_MM.Z must be greater than or equal to 0."),
      |                   ~~~~~~~~~~~~^~~~
Marlin\src\HAL\LPC1768\../../inc/../core/macros.h:260:34: note: in definition of macro 'CODE_3'
  260 | #define CODE_3( A,B,C,...) A; B; C
      |                                  ^
Marlin\src\HAL\LPC1768\../../inc/../core/macros.h:264:24: note: in expansion of macro '_CODE_N'
  264 | #define CODE_N(N,V...) _CODE_N(N,V)
      |                        ^~~~~~~
Marlin\src\HAL\LPC1768\../../inc/../core/types.h:43:32: note: in expansion of macro 'CODE_N'
   43 | #define LINEAR_AXIS_CODE(V...) CODE_N(LINEAR_AXES, V)
      |                                ^~~~~~
Marlin\src\HAL\LPC1768\../../inc/SanityCheck.h:1703:3: note: in expansion of macro 'LINEAR_AXIS_CODE'
 1703 |   LINEAR_AXIS_CODE(
      |   ^~~~~~~~~~~~~~~~
Marlin\src\HAL\LPC1768\../../inc/SanityCheck.h:1706:31: error: array subscript value '2' is outside the bounds of array type 'const float [2]'
 1706 |     static_assert(sbm[Z_AXIS] >= 0, "SENSORLESS_BACKOFF_MM.Z must be greater than or equal to 0."),
      |                   ~~~~~~~~~~~~^~~~
Marlin\src\HAL\LPC1768\../../inc/../core/macros.h:260:34: note: in definition of macro 'CODE_3'
  260 | #define CODE_3( A,B,C,...) A; B; C
[... 69 lines cut ...]
Marlin\src\HAL\LPC1768\../../inc/SanityCheck.h:1706:31: error: array subscript value '2' is outside the bounds of array type 'const float [2]'
 1706 |     static_assert(sbm[Z_AXIS] >= 0, "SENSORLESS_BACKOFF_MM.Z must be greater than or equal to 0."),
      |                   ~~~~~~~~~~~~^~~~
Marlin\src\HAL\LPC1768\../../inc/../core/macros.h:260:34: note: in definition of macro 'CODE_3'
  260 | #define CODE_3( A,B,C,...) A; B; C
      |                                  ^
Marlin\src\HAL\LPC1768\../../inc/../core/macros.h:264:24: note: in expansion of macro '_CODE_N'
  264 | #define CODE_N(N,V...) _CODE_N(N,V)
[... cut ...]
============================================================================ [FAILED] Took 8.42 seconds ============================================================================
Environment    Status    Duration
-------------  --------  ------------
LPC1769        FAILED    00:00:08.419
====================================================================== 1 failed, 0 succeeded in 00:00:08.419 ====================================================================== 
The terminal process "C:\Users\armin\.platformio\penv\Scripts\platformio.exe 'run'" terminated with exit code: 1.`

DerAndere1, 2021-06-05:
With the proposed fixes in PR 22056, you will need `#define SENSORLESS_BACKOFF_MM { 5, 5, 0 }` in your Configuration_adv.h in case you have that option enabled.

ellensp, 2021-06-06:
Fix has been merged. Closing.

closed 2021-06-06 by ellensp

slowbro, 2021-06-06:
I don't think that merge fixed the whole issue. With `{ 2, 2, 0 }` when only using sensorless homing on X and Y, I get...

```
In file included from Marlin/src/module/../inc/MarlinConfigPre.h:56,
                 from Marlin/src/module/../inc/MarlinConfig.h:28,
                 from Marlin/src/module/motion.h:31,
                 from Marlin/src/module/motion.cpp:27:
Marlin/src/module/motion.cpp: In function 'void homeaxis(AxisEnum)':
Marlin/src/module/../inc/../../Configuration_adv.h:743:42: error: too many initializers for 'const xy_float_t' {aka 'const XYval<float>'}
  743 | #define SENSORLESS_BACKOFF_MM  { 2, 2, 0 }     // (mm) Backoff from endstops before sensorless homing
      |                                          ^
Marlin/src/module/motion.cpp:1740:34: note: in expansion of macro 'SENSORLESS_BACKOFF_MM'
 1740 |       const xy_float_t backoff = SENSORLESS_BACKOFF_MM;
      |                                  ^~~~~~~~~~~~~~~~~~~~~
*** [.pio/build/LPC1768/src/src/module/motion.cpp.o] Error 1

```