MarlinFirmware/Marlin #22061: [BUG] AnyCubic I3Mega examples config

DamianLisu, 2021-06-06:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

I found big mistakes in examples config for AnyCubic i3 Mega.
This bug is from 2.0.8.1 and if you look in Configuration.h can you see:
line `146 #if TRIGORILLA_VERSION == 1`
`...`
and line `2605 #define TRIGORILLA_VERSION 0`

As you can see there is a condition first and then a variable setting.
This setting causes BLTouch to not work properly, because how is it supposed to work on a bad board

In old version was
`#ifndef MOTHERBOARD
  #define MOTHERBOARD BOARD_TRIGORILLA_14_11
#endif
`

### Version of Marlin Firmware

2.0.8.2

ellensp, 2021-06-06:
Yes the section 
```CPP
/** 
 * Select your version of the Trigorilla (RAMPS1.4) board here. 
 * 
 * 0 = Default Trigorilla 
 * 1 = Newer Trigorilla v1.1 (first seen late 2018) 
 * 
 * The only major difference is a slight change on the servo pin mapping. 
 * This setting only is relevant if you want to use BLtouch or similar 
 * mods to be used via servo pins. 
 * The new version is to be identified by a "TRIGORILLA1.1" lettering 
 * on the upper left of the PCB silkscreen. 
 */
#define TRIGORILLA_VERSION 0
```

Should be before 
```CPP
// Choose the name from boards.h that matches your setup
#if TRIGORILLA_VERSION == 1  
  #define MOTHERBOARD BOARD_TRIGORILLA_14_11
#else  
  #define MOTHERBOARD BOARD_TRIGORILLA_14
#endif
```

ellensp, 2021-06-06:
A PR has been created to fix this,

closed 2021-06-06 by ellensp