MarlinFirmware/Marlin #22062: [BUG]? home on Z_MAX not possible

Isengrimm, 2021-06-06:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

My printer uses a Prusa setup for the X axis, so I use two motors with separate drivers and separate end stops per driver.
The appropriate configuration for "#define Z_HOME_DIR -1" works correctly.

But now I would like to go to Z_MAX to align my portal. The actual bed height (Z0.00) should be queried with a probe (BLTouch) and G38.2.

Unfortunately, G34 is not an option for this printer, as the prerequisite for this is that the bed is always correctly leveled

### Bug Timeline

Since Marlin 2.0.8, 2.0.8.1 und 2.0.8.2

### Expected behavior

In the Configuration.h "#define Z_HOME_DIR -1" change to "1", compile, done

### Actual behavior

following error message:

Marlin\src\module\stepper.cpp: In static member function 'static void Stepper::pulse_phase_isr()':
Marlin\src\module\stepper.cpp:412:55: error: 'Z_HOME_TO_MIN' was not declared in this scope; did you mean 'X_HOME_TO_MIN'?
  412 |     #define Z_APPLY_STEP(v,Q) DUAL_ENDSTOP_APPLY_STEP(Z,v)
      |                                                       ^
Marlin\src\module\stepper.cpp:262:9: note: in definition of macro 'DUAL_ENDSTOP_APPLY_STEP'
  262 |     if (A##_HOME_TO_MIN) {                                                                                                  \
      |         ^
Marlin\src\module\stepper.cpp:1556:44: note: in expansion of macro 'Z_APPLY_STEP'
 1556 |     #define _APPLY_STEP(AXIS, INV, ALWAYS) AXIS ##_APPLY_STEP(INV, ALWAYS)
      |                                            ^~~~
Marlin\src\module\stepper.cpp:1572:9: note: in expansion of macro '_APPLY_STEP'
 1572 |         _APPLY_STEP(AXIS, !_INVERT_STEP_PIN(AXIS), 0); \
      |         ^~~~~~~~~~~
Marlin\src\module\stepper.cpp:1736:7: note: in expansion of macro 'PULSE_START'
 1736 |       PULSE_START(Z);
      |       ^~~~~~~~~~~
Marlin\src\module\stepper.cpp:412:55: error: 'Z_HOME_TO_MIN' was not declared in this scope; did you mean 'X_HOME_TO_MIN'?
  412 |     #define Z_APPLY_STEP(v,Q) DUAL_ENDSTOP_APPLY_STEP(Z,v)
      |                                                       ^
Marlin\src\module\stepper.cpp:262:9: note: in definition of macro 'DUAL_ENDSTOP_APPLY_STEP'
  262 |     if (A##_HOME_TO_MIN) {                                                                                                  \
      |         ^
Marlin\src\module\stepper.cpp:1556:44: note: in expansion of macro 'Z_APPLY_STEP'
 1556 |     #define _APPLY_STEP(AXIS, INV, ALWAYS) AXIS ##_APPLY_STEP(INV, ALWAYS)
      |                                            ^~~~
Marlin\src\module\stepper.cpp:1579:9: note: in expansion of macro '_APPLY_STEP'
 1579 |         _APPLY_STEP(AXIS, _INVERT_STEP_PIN(AXIS), 0); \
      |         ^~~~~~~~~~~
Marlin\src\module\stepper.cpp:1765:7: note: in expansion of macro 'PULSE_STOP'
 1765 |       PULSE_STOP(Z);

### Version of Marlin Firmware

Marlin 2.0.8.2 @platformIO

### Printer model

Anycubic i3 Mega

### Electronics

BTT SKR 1.4

### Your Slicer

Cura

### Host Software

SD Card (headless)

ellensp, 2021-06-06:
Please attach your configuration files.
I cannot reproduce from description alone.

Isengrimm, 2021-06-06:
Of course, if you need more please tell me

[Files_Isengrimm.zip](https://github.com/MarlinFirmware/Marlin/files/6604139/Files_Isengrimm.zip)

ellensp, 2021-06-06:
Your configs, once upgraded for current bugfix. Compiles without issues
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/6604221/Configuration.zip) for VERSION 02000900

Isengrimm, 2021-06-06:
i'm an idiot, i didn't pay attention to the difference between "latest release" and "bugfix ..." ... please excuse the confusion this caused. :-(

You are right, with the bugfix version 2.0.9 the bug is fixed and the compilation works without problems.

Thank you very much for your efforts and help!

closed 2021-06-06 by thisiskeithb