MarlinFirmware/Marlin #22063: [BUG] marlin crashes on printing from SD card if power loss recovery enabled

maheshhchandra, 2021-06-06:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

marlin crashes on printing from SD card if power loss recovery enabled
using marlin 2.0.8.2
SKR mini E3 V1.2

marlin auto build :
stm32f103rc_btt
stm32f103rc_btt_stm32
stm32f103rc_btt_512k
stm32f103rc_btt_512k_stm32

using BL touch
Filament runout sensor

PLR file is created on SD card but empty
attaching PLR file
[plr.zip](https://github.com/MarlinFirmware/Marlin/files/6604192/plr.zip)

### Bug Timeline

was working with marlin 2.0.7.2

### Steps to Reproduce

start print fron SD card if power loss recovery is switched on

### Version of Marlin Firmware

2.0.8.2

### Printer model

Ender 3 pro

### Electronics

BL touch, filament runout sensor

### Your Slicer

Cura

### Host Software

OctoPrint

### Additional information & file uploads

conf files
[conf.zip](https://github.com/MarlinFirmware/Marlin/files/6604196/conf.zip)

ellensp, 2021-06-06:
latest bugfix does even have 
stm32f103rc_btt_stm32
stm32f103rc_btt_512k_stm32
All stm32 env's where renamed

ellensp, 2021-06-06:
actual bugfix configs for testing VERSION 02000900 [Configs.zip](https://github.com/MarlinFirmware/Marlin/files/6604273/Configs.zip)

maheshhchandra, 2021-06-06:
For stable version
stm32f103rc_btt_stm32
stm32f103rc_btt_512k_stm32

For bugfix
stm32f103rc_btt_maple
stm32f103rc_btt_512k_maple

Both versions not working

ellensp, 2021-06-06:
that is interesting as stm32f103rc_btt_stm32 became stm32f103rc_btt 
and the original stm32f103rc_btt became stm32f103rc_btt_maple
So  according to this different build environments fail in stable vs bugfix

maheshhchandra, 2021-06-06:
I have checked all 4 non usb versions in both stable & bug fix
all crash just as I start print from SD

ellensp, 2021-06-06:
so your above statement is incorrect 
For bugfix the following fail
stm32f103rc_btt
stm32f103rc_btt_512k
stm32f103rc_btt_maple
stm32f103rc_btt_512k_maple

maheshhchandra, 2021-06-06:
yes all 4

ellensp, 2021-06-06:
Have you ted reformatting the sdcard, just in case it is a little corrupt only effecting new file creation

maheshhchandra, 2021-06-07:
Have tried 3 separate SD cards to rule out problem in SD cards
and also formatted the SD cards

thinkyhead, 2021-07-30:
Please test the [`bugfix-2.0.x` branch](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip) to see where it stands. If the problem has been resolved then we can close this issue. If the issue isn't resolved yet, then we should investigate further.

I expect this is still an issue because PLR is constantly getting broken by our extremely prolific contributors.

closed 2021-10-09 by github-actions[bot]