MarlinFirmware/Marlin #22065: [BUG] Can't build firmware due to missing file or directory - STM32F1

sarvenn, 2021-06-06:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

 [C:\users\sarve\.platformio\packages\framework-arduinoststm32-maple\STM32F1\variants\generic_stm32f103r\board.cpp] E:\Marlin-bugfix-2.0.x\.pio\build\STM32F103RC_btt_maple\.sconsign37.dblite: No such file or directory

I tried all the environments except USB one. All four have failed. Terminal text files are attached.
[STM32F103RC_btt.txt](https://github.com/MarlinFirmware/Marlin/files/6604257/STM32F103RC_btt.txt)
[STM32F103RC_btt_512K.txt](https://github.com/MarlinFirmware/Marlin/files/6604258/STM32F103RC_btt_512K.txt)
[STM32F103RC_btt_maple.txt](https://github.com/MarlinFirmware/Marlin/files/6604259/STM32F103RC_btt_maple.txt)
[STM32F103RC_btt_512K_maple.txt](https://github.com/MarlinFirmware/Marlin/files/6604260/STM32F103RC_btt_512K_maple.txt)

### Bug Timeline

new

### Expected behavior

Successfully built firmware files

### Actual behavior

no success

### Steps to Reproduce

1. select the environement from auto build marlin and click on build

### Version of Marlin Firmware

Latest bugfix2.0 as of today

### Printer model

Ender 3 pro

### Electronics

skr mini e3 v2

ellensp, 2021-06-06:
you have to give platformio a moment to copy files and build databases
Just try again in a few moments

ellensp, 2021-06-06:
Either that or something is really broken in your platformio or python version

closed 2021-06-06 by ellensp

DerAndere1, 2021-06-12:
Sometimes it helps to do a clean (In the left sidebar klick PlatformIO -> Project Tasks -> Default -> General -> Clean All)
and to delete the hidden .pio/libdeps folder
Then wait a minute and build again