MarlinFirmware/Marlin #22067: Filament Runout Compilation error and missing larm_cortexM4lf_math

FabioMC, 2021-06-06:
### Did you test the latest `bugfix-2.0.x` code?

No, but I will test it now!

### Bug Description

First time compiling 2.0.8.2. Before I was using version 2.0.7.2 with no issues during compilation. Right now I am receiving two erros that crashes the whole process:

The first issue is related to Runout Filmament script I guess.

Compiling .pio\build\mks_robin_nano_v3_usb_flash_drive_msc\src\src\gcode\bedlevel\G42.cpp.o
Compiling .pio\build\mks_robin_nano_v3_usb_flash_drive_msc\src\src\gcode\bedlevel\M420.cpp.o
In file included from C:\Users\fabio\.platformio\packages\framework-arduinoststm32\cores\arduino/WString.h:29,
                 from C:\Users\fabio\.platformio\packages\framework-arduinoststm32\cores\arduino/Print.h:26,
                 from C:\Users\fabio\.platformio\packages\framework-arduinoststm32\cores\arduino/Stream.h:26,
                 from C:\Users\fabio\.platformio\packages\framework-arduinoststm32\cores\arduino/HardwareSerial.h:29,
                 from C:\Users\fabio\.platformio\packages\framework-arduinoststm32\cores\arduino/WSerial.h:5,
                 from C:\Users\fabio\.platformio\packages\framework-arduinoststm32\cores\arduino/wiring.h:47,
                 from C:\Users\fabio\.platformio\packages\framework-arduinoststm32\cores\arduino/Arduino.h:36,
                 from Marlin\src\feature\../sd/../inc/../HAL/./STM32/../shared/Marduino.h:36,
                 from Marlin\src\feature\../sd/../inc/../HAL/./STM32/HAL.h:28,
                 from Marlin\src\feature\../sd/../inc/../HAL/HAL.h:30,
                 from Marlin\src\feature\../sd/../inc/MarlinConfig.h:31,
                 from Marlin\src\feature\../sd/cardreader.h:24,
                 from Marlin\src\feature\runout.h:28,
                 from Marlin\src\feature\runout.cpp:31:
Marlin\src\feature\runout.cpp: In function 'void event_filament_runout(uint8_t)':
Marlin\src\feature\../inc/../../Configuration.h:1340:34: warning: too many arguments for format [-Wformat-extra-args]
 1340 |   #define FILAMENT_RUNOUT_SCRIPT "M600"
C:\Users\fabio\.platformio\packages\framework-arduinoststm32\cores\arduino/avr/pgmspace.h:71:40: note: in definition of macro 'sprintf_P'
   71 | #define sprintf_P(s, ...) sprintf((s), __VA_ARGS__)
      |                                        ^~~~~~~~~~~
Marlin\src\feature\runout.cpp:130:25: note: in expansion of macro 'PSTR'
  130 |       sprintf_P(script, PSTR(FILAMENT_RUNOUT_SCRIPT), tool);
      |                         ^~~~
Marlin\src\feature\runout.cpp:130:30: note: in expansion of macro 'FILAMENT_RUNOUT_SCRIPT'
  130 |       sprintf_P(script, PSTR(FILAMENT_RUNOUT_SCRIPT), tool);
      |                              ^~~~~~~~~~~~~~~~~~~~~~
Compiling .pio\build\mks_robin_nano_v3_usb_flash_drive_msc\src\src\gcode\bedlevel\abl\G29.cpp.o
Compiling .pio\build\mks_robin_nano_v3_usb_flash_drive_msc\src\src\gcode\bedlevel\abl\M421.cpp.o

The second is an error that hapens at the very end of the compilation process and I believe it is related to platformIO.

Linking .pio\build\mks_robin_nano_v3_usb_flash_drive_msc\firmware.elf
c:/users/fabio/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld.exe: cannot find -larm_cortexM4lf_math   
collect2.exe: error: ld returned 1 exit status
*** [.pio\build\mks_robin_nano_v3_usb_flash_drive_msc\firmware.elf] Error 1

Is anyone faced some similar problem before? Any hint to fix those errors?

Thanks

### Version of Marlin Firmware

2.0.8.2

rhapsodyv, 2021-06-06:
A missing lib while building seems a corrupted PlatformIO install… 🤔

FabioMC, 2021-06-06:
I tried to unistall PlatformIO and install it again but did not work. Both issues remains.

thisiskeithb, 2021-06-06:
Please attach your configs.

FabioMC, 2021-06-06:
Here they are

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/6604758/Configuration.zip)

thisiskeithb, 2021-06-06:
Try deleting the hidden `C:\Users\fabio\.platformio\` and `Marlin\.pio\` folders and reinstall PIO again.

<details><summary>There were some warnings:</summary>

```prolog
Compiling .pio/build/mks_robin_nano_v3_usb_flash_drive_msc/src/src/gcode/config/M220.cpp.o
In file included from /Users/keith/.platformio/packages/framework-arduinoststm32/cores/arduino/WString.h:29,
                 from /Users/keith/.platformio/packages/framework-arduinoststm32/cores/arduino/Print.h:26,
                 from /Users/keith/.platformio/packages/framework-arduinoststm32/cores/arduino/Stream.h:26,
                 from /Users/keith/.platformio/packages/framework-arduinoststm32/cores/arduino/HardwareSerial.h:29,
                 from /Users/keith/.platformio/packages/framework-arduinoststm32/cores/arduino/WSerial.h:5,
                 from /Users/keith/.platformio/packages/framework-arduinoststm32/cores/arduino/wiring.h:47,
                 from /Users/keith/.platformio/packages/framework-arduinoststm32/cores/arduino/Arduino.h:36,
                 from Marlin/src/feature/../sd/../inc/../HAL/./STM32/../shared/Marduino.h:36,
                 from Marlin/src/feature/../sd/../inc/../HAL/./STM32/HAL.h:28,
                 from Marlin/src/feature/../sd/../inc/../HAL/HAL.h:30,
                 from Marlin/src/feature/../sd/../inc/MarlinConfig.h:31,
                 from Marlin/src/feature/../sd/cardreader.h:24,
                 from Marlin/src/feature/runout.h:28,
                 from Marlin/src/feature/runout.cpp:31:
Marlin/src/feature/runout.cpp: In function 'void event_filament_runout(uint8_t)':
Marlin/src/feature/../inc/../../Configuration.h:1340:34: warning: too many arguments for format [-Wformat-extra-args]
 1340 |   #define FILAMENT_RUNOUT_SCRIPT "M600"
/Users/keith/.platformio/packages/framework-arduinoststm32/cores/arduino/avr/pgmspace.h:71:40: note: in definition of macro 'sprintf_P'
   71 | #define sprintf_P(s, ...) sprintf((s), __VA_ARGS__)
      |                                        ^~~~~~~~~~~
Marlin/src/feature/runout.cpp:130:25: note: in expansion of macro 'PSTR'
  130 |       sprintf_P(script, PSTR(FILAMENT_RUNOUT_SCRIPT), tool);
      |                         ^~~~
Marlin/src/feature/runout.cpp:130:30: note: in expansion of macro 'FILAMENT_RUNOUT_SCRIPT'
  130 |       sprintf_P(script, PSTR(FILAMENT_RUNOUT_SCRIPT), tool);
      |                              ^~~~~~~~~~~~~~~~~~~~~~
```
</details>

<details><summary>But it builds fine:</summary>

```prolog
Checking size .pio/build/mks_robin_nano_v3_usb_flash_drive_msc/firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [=====     ]  47.9% (used 62828 bytes from 131072 bytes)
Flash: [===       ]  34.3% (used 359152 bytes from 1048576 bytes)
Building .pio/build/mks_robin_nano_v3_usb_flash_drive_msc/firmware.bin
noencrypt([".pio/build/mks_robin_nano_v3_usb_flash_drive_msc/firmware.bin"], [".pio/build/mks_robin_nano_v3_usb_flash_drive_msc/firmware.elf"])
====================================================================================================================== [SUCCESS] Took 55.79 seconds ======================================================================================================================

Environment                            Status    Duration
-------------------------------------  --------  ------------
mks_robin_nano_v3_usb_flash_drive_msc  SUCCESS   00:00:55.794
======================================================================================================================= 1 succeeded in 00:00:55.794 =======================================================================================================================

Terminal will be reused by tasks, press any key to close it.
```
</details>

---

This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- [Marlin Documentation](https://marlinfw.org) hosted on MarlinFW.org
- [Marlin Forum](https://forums.reprap.org/list.php?415) hosted on RepRap.org
- [Marlin Firmware](https://www.facebook.com/groups/1049718498464482/) Facebook Group 
- [Marlin Firmware for 3D Printers](https://www.facebook.com/groups/3Dtechtalk/) Facebook Group
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- [Marlin Firmware](https://discord.gg/n5NJ59y) Discord server

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2021-06-06 by thisiskeithb

FabioMC, 2021-06-06:
Thanks. It worked. Although I still have the Filament Runout warnings the firmware concluded and looks like its working fine.