MarlinFirmware/Marlin #22071: [BUG] Changing Z-Offset while printing is not working

DasSmeagle, 2021-06-07:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

When changing "Z-offset" in "Tune" menu while printing "nothing happens". Have to stop printing, change it and restart print.

This worked fine on the creality firmware I used before, but it did not work on any Marlin I tried. I tried 2.0.x, bugfix-2.0.x and Release 2.0.8.2.

Is this something I have to enable in config? But if it is, I still consider this a bug: If I CAN change the Z-Offset in Tune menu, I would expect it to have itt change while printing.

I am Using Ende-3 V2 with BLTouch, bilinear bed leveling.

### Bug Timeline

not sure, saw it in several versions and it is my frist use

### Expected behavior

When able to select "Z-Offset" I expect it to change while already printing. 

### Actual behavior

I can set Z-Offset to extreme values and see no difference in printing.

### Steps to Reproduce

- Use BLTouch-Bed-Leveling and Auto-Home using BLtouch für Z axis
- start printing
- go to "Tune" menu while printing
- change z-offset to some different value (like from -2 to +5 -> should be VERY visible)
- "Nothing" happens

### Version of Marlin Firmware

branch 2.0.8.2

### Printer model

Ender-3 V2

### Electronics

stock (4.2.2 Board)

### Add-ons

BLTouch (from Creality)

### Your Slicer

Cura

### Host Software

SD Card (headless)

### Additional information & file uploads

[Marlin_Config_Oliver.zip](https://github.com/MarlinFirmware/Marlin/files/6608214/Marlin_Config_Oliver.zip)

I could do a video, but I think it is clear enough?

DasSmeagle, 2021-06-11:
Any input on this? I cannot find the reason why this is not working for me. I am using BLTouch and after changing Z-offset I have to "Auto Home" before it gets active.

Tried to use "Babysteps" feature, but I cannot see anything, guess it is not compatible with my Display (DWIN)?

DerAndere1, 2021-06-12:
If you connect your printer to a PC that runs a host software like printrun or repetier host, does it have an effect if you adjust Z home offset by sending G-code like `M206 Z20` from the host software?

DasSmeagle, 2021-06-12:
Al lot of stuff is not working, I guess the DWIN of Ender-3 V2 is not really supported in this firmware. So I switched to another branch of Marlin, " Jyers - Marlin - Ender 3 V2 Extensible UI" which is specialized for my printer. My problem is solved for now. I used my config in that branch and it worked fine - so it was not an error in my config.

I hope, support for DWIN will be better in the future...

thisiskeithb, 2021-06-13:
> I hope, support for DWIN will be better in the future...

It will be. There's an open PR for MarlinUI support for Crealtiy DWIN LCDs: https://github.com/MarlinFirmware/Marlin/pull/19371

mriscoc, 2021-06-14:
> Al lot of stuff is not working, I guess the DWIN of Ender-3 V2 is not really supported in this firmware. So I switched to another branch of Marlin, " Jyers - Marlin - Ender 3 V2 Extensible UI" which [...]

Strange because Jyers and Me, we both are using the bugfix branch as base. I will try to do more research in the week.

DasSmeagle, 2021-06-14:
I can only say, a lot of options don't work on This branch but work on the Jyers branch. I guess it has to do with the DWIN not enabling "HAS_LCD_MENU".  But just guessing. :) I just see, a lot of options don't work on DWIN, like Babystepping, changing Z-Offset, adding "Home" menu items etc. and they all work on the Jyers branch.

YAost, 2021-07-04:
Enabling BABYSTEP_ZPROBE_OFFSET in the BABYSTEPPING section of Configuration_adv.h helped me, you could try that as well.
I believe that in the earlier versions (or at least with LCD on other printers) this option wasn't required for live offset changing, but not sure.

thinkyhead, 2021-07-30:
If you want the probe Z offset to take effect in the current session you need to use `BABYSTEP_ZPROBE_OFFSET`. Otherwise you must do `G28` (and if needed, `G29`) to do a new homing and make a new mesh using the new offset. It might be possible to do some clever things like modifying the whole mesh and your current reckoned Z position based on the modified Z Probe Offset. But it gets a little complicated in the various interactions between logical and stepper positions, and there is a Catch-22 situation when modifying the Z Probe Offset while also babystepping.

Anyone who decides to take the plunge and make the attempt will run into these complications, and it may be quite frustrating. Moreover, all such changes which affect the user interface have to be applied across all the display implementations, which is a further annoyance.

So, if this is very much desired, you should spend a lot of time on the Marlin Discord and solicit help.

closed 2021-10-09 by github-actions[bot]