MarlinFirmware/Marlin #22073: [FR] Use Z-Stop-Switch as "failsafe" on Auto-Homeing when BLTouch Meassuring fails

DasSmeagle, 2021-06-07:
### Is your feature request related to a problem? Please describe.

My BLTouch (and according to my result when searching for this problem I am not alone) will sometimes fail to meassure/retract.

See this video at 1:40 :

https://www.youtube.com/watch?v=eIPW7iRT3wU

It happens with a lot of different version of Firmware I tried, so I think it is a hardware problem with the sensor.

It happens not very often, about 1 in 100 probings.

When it happens, the BLTouch starts blinking, the Bed Leveling is aborted.
It can happen in Auto-Homing, too, but in that case it is worse, the nozzle will crash into the bed when it happens.

### Are you looking for hardware support?

Ender-3 V2 with Creality BLTouch package

### Describe the feature you want

In my system the Z-Stop-Switch is pressed a few mm later then the BLTouch registering the touch of bed.

So it would be possible to STOP the head when triggering the Z-Stop-Switch.

So I suggest a configurable option to use Z-Stop-Switch as a "safety stop" when using probe to auto home. If it makes sense in some configs maybe in Bed Leveling, too, but on my config that wouldn't work as the switch is not triggerd in that case.
[Marlin_Config_Oliver.zip](https://github.com/MarlinFirmware/Marlin/files/6608336/Marlin_Config_Oliver.zip)

slowbro, 2021-06-10:
A potential problem with this is that generally the BLTouch is electrically connected to the same place as the switch, so it would not be possible to know which triggered it. If you had a spare digital input, that would be do-able.

I wonder if there is some way for Marlin to know when the BLTouch is in that state, and halt?

DasSmeagle, 2021-06-10:
> A potential problem with this is that generally the BLTouch is electrically connected to the same place as the switch, so it would not be possible to know which triggered it. If you had a spare digital input, that would be do-able.

I know, SOME people connect the BLTouch to the 2pin connector of Z-End-Stop-Switch and only 3pins to the BLTocuh connector, yes, but I don't know why... The original BLTouch Set from creality contains a 5 pin cable and connects to the seperate 5 pinbltouch connector on the board.

It may be because in the default config Marlin is configered to use the Z-Stop-Switch connector, which is not how I connected it, so I changed it in config.

In this video the Z-End-Switch is still connected to its normal port and I can press or hold it without interfering the probing:

https://www.youtube.com/watch?v=EG_DvuNiJRs

thinkyhead, 2021-07-30:
Currently the `Z_PROBE_LOW_POINT` is the "failsafe" but it may be possible to throw "probing failed" if the Z endstop triggers in setups where the Probe and the Z MIN endstop are both in use. This requires the probe pin to be separate from the Z MIN pin, which is not unusual, as we have the option to not home with the probe. It would require a bunch of finagling in an area of the code that right at this moment we are only debugging, and not modifying. So, check back in a few weeks after we have a bunch of the current bugs solved and the bug reports seem to be slowing down.

closed 2026-11-12 by thinkyhead