MarlinFirmware/Marlin #22074: [BUG] Change baud using M575 causes BLtouch to fail deploy on next G28 resulting in nozzle crashing into bed

badbod, 2021-06-07:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

Change baud from 115200 to 25000 using M575 results in BLtouch failing to deploy probe on next G28. resulting in hot nozzle on bed action.

### Bug Timeline

First noticed in 2.7 but didnt realise what caused it.

### Expected behavior

Expect BLtouch to be unaffected or reset by serial port baud rate changes.

### Actual behavior

Probe fails to deploy with resultant nozzle 'plunge of death' into bed.

### Steps to Reproduce

Use M575 B250000 gcode command to change serial port baud rate then auto home with LCD menu or G28 .

### Version of Marlin Firmware

2.8.2

### Printer model

Ender 3 V2

### Electronics

2 off 4.2.7 upgrade controller boards.

### Add-ons

BLtouch

### Host Software

OctoPrint

### Additional information & file uploads

[configs.zip](https://github.com/MarlinFirmware/Marlin/files/6609355/configs.zip)

Same problem with stock configs apart from enable M575 command.
Same if using pronterface terminal to issue M575.
There is no UART mod for SPREAD toggle and same result on unmodified STEALTHCHOP board

Changing the Baud to 250000 in firmware and flash has no problems.  It seems to be the change of baud, reconnect and G28 that does it.

Has done it a few times over last few months with various versions of boards and firmware versions. Only just worked out what was actually causing it.

foxylion, 2021-06-14:
I have a similar issue, after calling `M575 P0 B250000` the printer fails do deploy the BLTouch on future commands. And the steppers (x, y and z) begin to make some weird (loud) noise when homing.

badbod, 2021-06-27:
> I have a similar issue, after calling `M575 P0 B250000` the printer fails do deploy the BLTouch on future commands. And the steppers (x, y and z) begin to make some weird (loud) noise when homing.

can't say I noticed any extra noise, also have not tried to repeat using 2.0.9/2.0.9.1, and I have changed board and added lots steppers and stuff, I am about to build 2.0.9.1 so I will report back in a day or two if the problem still persists.

badbod, 2021-07-03:
due to explosive popularity of this bug report, and the fact I am hesitant to reproduce on my shiny upgrades, probably best to close this, just remember to not test baud rates by issuing M575. :-)

closed 2021-07-03 by badbod