MarlinFirmware/Marlin #22076: When Extruders = 0 , six axis config sanity checks fail. Laser Cutters do not have extruders.

descipher, 2021-06-07:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

Bad Sanity check, should understand that Laser cutters do not have 4 axis.  

### Bug Timeline

Close to Jun 6th 2021

### Expected behavior

Sanity should check if LASER_FEATURE is defined before assuming this target is a CNC Machine. 

### Actual behavior

Fails to compile a valid laser config.

### Steps to Reproduce

Set Extruders to 0

### Version of Marlin Firmware

bugfix-2.0.x

descipher, 2021-06-07:
Marlin\src\HAL\AVR\../../inc/SanityCheck.h:574:1: error: static assertion failed: AXIS_RELATIVE_MODES must contain X Y Z elements.
 static_assert(COUNT(arm) == LOGICAL_AXES, "AXIS_RELATIVE_MODES must contain " _LOGICAL_AXES_STR "elements.");

descipher, 2021-06-07:
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/6611107/Configuration.zip)

thisiskeithb, 2021-06-07:
@DerAndere1: More 6 axes follow-up if you have a chance.

DerAndere1, 2021-06-07:
As the sanity check tells you, the array AXIS_RELATIVE_MODES needs to have elemets for X,Y, and Z in your case. That means exactly 3 elements, so you have to change it to
```
#define AXIS_RELATIVE_MODES { false, false, false }
```

Background:  AXIS_RELATIVE_MODES must have one element for each so called logical axis (internally called LINEAR_AXES). Your config has LINEAR_AXES unspecified, that means it falls back to LINEAR_AXES 3. For a machine with extruders (no matter how many),  LOGICAL_AXES = LINEAR_AXES + 1.
But you have EXTRUDERS 0, that means LOGICAL_AXES = LINEAR_AXES, so you have 3 logical axes. 

We need to update the documentation. 
@thinkyhead : is the documentation supposed to reflect the last tagged release or current bugfix-2.0.x ?

descipher, 2021-06-07:
Just a thought on this, extruders are not an AXIS so AXIS_RELATIVE_MODE has no meaning to the user. E.g.  the definition of the define with AXIS_RELATIVE_MODES { false, false, false, false } does not describe an extruder or axis. It appears to be an bool array describing motor movement mode which is absolute or relative positioning. Therefore it could be a dynamic  array and default to false if not defined which could avoid this issue. 

#define X_AXIS_MOVEMENT_MODE = ABS 

Could have more clarity and applied to the dynamic array element is required.

descipher, 2021-06-08:
Should we close this issue? It’s not a bug. We could have some additional comments or documentation to handle the user defines though. Also could approach it with the dynamic method option to simplify configuration items for the users.

thisiskeithb, 2021-06-08:
> Should we close this issue? It’s not a bug. We could have some additional comments or documentation to handle the user defines though. Also could approach it with the dynamic method option to simplify configuration items for the users.

Leaving it open for discussion on how to proceed works.

thinkyhead, 2021-07-30:
@DerAndere1 — Aha, your answer above is actually correct! 😉 The sanity-check error points to the fix, even if it isn't perfectly clear. It is in the same idiom as the rest of the array-validating sanity checks.

closed 2021-07-30 by thinkyhead