MarlinFirmware/Marlin #22082: [BUG] MKS robin nano v.2, MKSUI, Compilation error

Malderin, 2021-06-08:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

Board: MKS robin nano v.2
Last marlin 2.x bugfix
[Configs.zip](https://github.com/MarlinFirmware/Marlin/files/6618510/Configs.zip)

Compilation error:

```
Compiling .pio\build\mks_robin_nano35\src\src\HAL\STM32\usb_host.cpp.o
In file included from Marlin\src\HAL\STM32\../../inc/MarlinConfigPre.h:37,
                 from Marlin\src\HAL\STM32\../../inc/MarlinConfig.h:28,
                 from Marlin\src\HAL\STM32\timers.cpp:24:
Marlin\src\HAL\STM32\timers.cpp: In function 'void SetTimerInterruptPriorities()':
Marlin\src\HAL\STM32\timers.cpp:190:44: error: 'setInterruptPriority' is not a member of 'SoftwareSerial'
  190 |   TERN_(HAS_TMC_SW_SERIAL, SoftwareSerial::setInterruptPriority(SWSERIAL_TIMER_IRQ_PRIO, 0));
      |                                            ^~~~~~~~~~~~~~~~~~~~
Marlin\src\HAL\STM32\../../inc/../core/macros.h:558:26: note: in definition of macro 'THIRD'
  558 | #define THIRD(a,b,c,...) c
      |                          ^
Marlin\src\HAL\STM32\../../inc/../core/macros.h:204:29: note: in expansion of macro '___TERN'
  204 | #define __TERN(T,V...)      ___TERN(_CAT(_NO,T),V)  // Prepend '_NO' to get '_NOT_0' or '_NOT_1'
      |                             ^~~~~~~
Marlin\src\HAL\STM32\../../inc/../core/macros.h:203:29: note: in expansion of macro '__TERN'
  203 | #define _TERN(E,V...)       __TERN(_CAT(T_,E),V)    // Prepend 'T_' to get 'T_0' or 'T_1'
      |                             ^~~~~~
Marlin\src\HAL\STM32\../../inc/../core/macros.h:202:29: note: in expansion of macro '_TERN'
  202 | #define TERN_(O,A)          _TERN(_ENA_1(O),,A)     // OPTION ? 'A' : '<nul>'
      |                             ^~~~~
Marlin\src\HAL\STM32\timers.cpp:190:3: note: in expansion of macro 'TERN_'
  190 |   TERN_(HAS_TMC_SW_SERIAL, SoftwareSerial::setInterruptPriority(SWSERIAL_TIMER_IRQ_PRIO, 0));
      |   ^~~~~
Compiling .pio\build\mks_robin_nano35\src\src\HAL\STM32\usb_serial.cpp.o
*** [.pio\build\mks_robin_nano35\src\src\HAL\STM32\timers.cpp.o] Error 1
========================= [FAILED] Took 97.58 seconds =========================
 
Environment       Status    Duration
----------------  --------  ------------
mks_robin_nano35  FAILED    00:01:37.576
==================== 1 failed, 0 succeeded in 00:01:37.576 ====================
```

### Version of Marlin Firmware

Marlin 2.0 bugfix (latest on 08.06.2021)

ellensp, 2021-06-08:
I cannot replicate this error.

I get a different error

```
In file included from Marlin/src/lcd/extui/mks_ui/draw_ui.h:85,
                 from Marlin/src/MarlinCore.cpp:72:
Compiling .pio/build/mks_robin_nano35/src/src/feature/tmc_util.cpp.o
Marlin/src/lcd/extui/mks_ui/wifiSerial.h:38:10: fatal error: libmaple/libmaple_types.h: No such file or directory
   38 | #include <libmaple/libmaple_types.h>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.

```
this is due to MKS_WIFI_MODULE, which has not been upgraded away from the old maple code. 
Ie it only works under mks_robin_nano35_maple build environment at this time.

ellensp, 2021-06-08:
You probably need to delete the hidden .pio directory in your marlin folder.
If you built under the older mks_robin_nano35 then upgraded marlin, platformio doesn't notice that mks_robin_nano35 is not the same as it was and joins old and new build environments. 
Also a build clean will not remove enough, you need to delete the .pio folder

Malderin, 2021-06-09:
> I cannot replicate this error.
>
> I get a different error
>
> ```
> In file included from Marlin/src/lcd/extui/mks_ui/draw_ui.h:85,
>                  from Marlin/src/MarlinCore.cpp:72:
> Compiling [...]

If you specify:
#define X_DRIVER_TYPE  TMC2209
#define Y_DRIVER_TYPE  TMC2209
#define Z_DRIVER_TYPE  TMC2209
#define E0_DRIVER_TYPE TMC2209
 you will get an error like I have above.

With a4988 I also get an error:

```
In file included from Marlin\src\lcd/extui/mks_ui/draw_ui.h:85,
                 from Marlin\src\MarlinCore.cpp:72:
Marlin\src\lcd/extui/mks_ui/wifiSerial.h:38:10: fatal error: libmaple/libmaple_types.h: No such file or directory
   38 | #include <libmaple/libmaple_types.h>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
```

Malderin, 2021-06-09:
> You probably need to delete the hidden .pio directory in your marlin folder.
> If you built under the older mks_robin_nano35 then upgraded marlin, platformio doesn't notice that mks_robin_nano35 is [...]

Tried it, it doesn't help.

ellensp, 2021-06-09:
Using your provided files, without modification (ie they have #define X_DRIVER_TYPE  TMC2209, #define Y_DRIVER_TYPE  TMC2209, #define Z_DRIVER_TYPE  TMC2209 etc)

I can not reproduce your error

Malderin, 2021-06-09:
> Using your provided files, without modification (ie they have #define X_DRIVER_TYPE TMC2209, #define Y_DRIVER_TYPE TMC2209, #define Z_DRIVER_TYPE TMC2209 etc)
> 
> I can not reproduce your error

I updated the comment above, you're right.

ellensp, 2021-06-11:
I believe  this PR will fix this https://github.com/MarlinFirmware/Marlin/pull/22109

Also this is now just a  Duplicate of https://github.com/MarlinFirmware/Marlin/issues/21069

So I'm closing it.

Malderin, 2021-06-11:
> I believe this PR will fix this #22109

With this PR a new error:

```
Compiling .pio\build\mks_robin_nano35\src\src\lcd\extui\mks_ui\tft_lvgl_configuration.cpp.o
Marlin\src\lcd\extui\mks_ui\tft_lvgl_configuration.cpp:49:12: fatal error: ../../../../module/servo.h: No such file or directory
   49 |   #include "../../../../module/servo.h"
      |            ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
Compiling .pio\build\mks_robin_nano35\src\src\lcd\extui\mks_ui\tft_multi_language.cpp.o
*** [.pio\build\mks_robin_nano35\src\src\lcd\extui\mks_ui\tft_lvgl_configuration.cpp.o] Error 1
 
========================= [FAILED] Took 250.04 seconds =========================
Environment       Status    Duration
----------------  --------  ------------
mks_robin_nano35  FAILED    00:04:10.038
==================== 1 failed, 0 succeeded in 00:04:10.038 ====================
```