MarlinFirmware/Marlin #22084: [BUG] Non-updated value for MARLIN_HEX_VERSION

DarwinGeez, 2021-06-09:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

Found a bug in file Marlin/src/inc/Version.h, as it had the wrong version number
Line #55 presently is:
#define MARLIN_HEX_VERSION 02000801

It should be:
#define MARLIN_HEX_VERSION 02000900
In order to align with the rest of the files in the 6/8/2021 point release of Marlin 2.0.8.2

With this small change, the code compiles perfectly for me.
So it's a very simple bug.
I have a Creality Ender 3 Pro with the v4.2.7 mainboard.

Thanks!

### Bug Timeline

With 6/8/2021 point release of Marlin 2.0.8.2

### Expected behavior

Clean compile without errors

### Actual behavior

Had compile errors that looked like:
(excerpt of output)
…
Compiling .pio/build/STM32F103RET6_creality/src/src/HAL/STM32F1/HAL_SPI.cpp.o
Compiling .pio/build/STM32F103RET6_creality/src/src/HAL/STM32F1/MarlinSerial.cpp.o
Compiling .pio/build/STM32F103RET6_creality/src/src/HAL/STM32F1/SPI.cpp.o
In file included from Marlin/src/HAL/STM32F1/../../inc/MarlinConfig.h:49:0,
                 from Marlin/src/HAL/STM32F1/HAL.cpp:30:
Marlin/src/HAL/STM32F1/../../inc/SanityCheck.h:74:4: error: #error "Your Configuration.h file is for a newer version of Marlin. Upgrade Marlin or downgrade your Configuration.h."
   #error "Your Configuration.h file is for a newer version of Marlin. Upgrade Marlin or downgrade your Configuration.h."
    ^~~~~
In file included from Marlin/src/HAL/STM32F1/../../inc/MarlinConfig.h:49:0,
                 from Marlin/src/HAL/STM32F1/HAL.cpp:30:
Marlin/src/HAL/STM32F1/../../inc/SanityCheck.h:79:4: error: #error "Your Configuration_adv.h file is for a newer version of Marlin. Upgrade Marlin or downgrade your Configuration_adv.h."
   #error "Your Configuration_adv.h file is for a newer version of Marlin. Upgrade Marlin or downgrade your Configuration_adv.h."
    ^~~~~
(etc…)

### Steps to Reproduce

Compile it for Creality Ender 3 Pro with 427 mainboard
although the bug is generic to the specific printer

### Version of Marlin Firmware

2.0.8.2

### Printer model

Creality Ender 3 Pro with 427 mainboard

### Your Slicer

Cura

### Host Software

SD Card (headless)

ellensp, 2021-06-09:
bugfix config have #define CONFIGURATION_H_VERSION 02000900
https://github.com/MarlinFirmware/Configurations/blob/bugfix-2.0.x/config/examples/Creality/Ender-3%20Pro/CrealityV427/Configuration.h#L40

Which matches #define MARLIN_HEX_VERSION 02000900
ib marlin bugfix https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.0.x/Marlin/src/inc/Version.h#L55

If your Version.h says define MARLIN_HEX_VERSION 02000801
you are not running bugfix code, and should not be using bugfix Config files

closed 2021-06-09 by ellensp

thisiskeithb, 2021-06-09:
> If your Version.h says define MARLIN_HEX_VERSION 02000801
> you are not running bugfix code, and should not be using bugfix Config files

@ellensp: This was intentional. See https://github.com/MarlinFirmware/Configurations/pull/509#issuecomment-855177682

ellensp, 2021-06-09:
@thisiskeithb 

they say it is still in bugfix, which it clearly is not. 

in marlin bugfix https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.0.x/Marlin/src/inc/Version.h#L55
#define MARLIN_HEX_VERSION 02000900

They are not taking about 2.0.8.2 vs 2.0.8.1

thisiskeithb, 2021-06-09:
> they say it is still in bugfix, which it clearly is not.

Oh. That changed after my PRs were closed.