MarlinFirmware/Marlin #22088: [BUG] (Unable to build for Anycubic Mega Zero 2.0)

Diono, 2021-06-09:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

I try to run the build for Marlin bugfix-2.0.x config by Dust, Anycubic Mega 2.0
but each melzi environments failed to build

i have more of 5 errors like
```
c:\users\diono\.platformio\packages\toolchain-atmelavr\avr\include\avr\iom1284p.h:1:8: error: stray '\211' in program
 \x8d☻  H‰†`☺  Hƒ¦h☺   A÷ ►  t◄A‹Ö3ÉèèüòÿH‰†°   H‹†ˆ   H‰†0☺  A‹☼ºà♫riL‹l$p☼ºà
☼‚\x81   H‹\$8H‹ÓH‹Îè£♂òÿ¸☻   f‰G↕H‹ÎèŠ$  ‹A+D$►A9D$↑s^¹÷♠  èar÷ÿA‹è ♦  ‹ÓH‹ÎèjèþÿH‹œ$€   é?ÿÿÿH\x8d♥  D‰l$ L‹l$pM‹ÍL‹ÃH‹Îè>ÒÿÿA‹éuÿÿÿH‹Îè²:ÿÿérÿÿÿH‹Îèõ☺  ëW‹Ø‰„$˜   ‰„$►☺  =♣  Àt§=☻  €t♫=÷♠  t=Æ♠  u♀»÷♠  ‰œ$►☺  H‹D$hfƒH►♦‰X↑H‹L$xè¶†øÿ‹Ëè­q÷ÿ\x90A÷ @  ☼…©   fD‰w↕è%¥öÿH…À☼„§   H‹HXH‰Œ$°   H‰Œ$ˆ   H‰pXDˆ´$\x90   H‹D$`H‹@ ☼·Œ$►☺  H…À☼…„   I‹D$8H…ÀH‹„$    H‹♀ÈHc‡t☻  HÁè♥‰D$@…Àt A÷  u!D‰t$4D‰t$0E☼·G↕D‹ÈI‹Õè†šùÿƒd$0 ëpA+Æ‰D$@ëÖH‹\x8fÐ☻  è’FÿÿéFÿÿÿèPCõÿH…À☼…Kÿÿÿ\x8dH♫èÏp÷ÿÌH‹♦ÈH…À☼„oÿÿÿD‰t$4D‰t$0H‹Îÿ§▬@☺ ë¦‰D$T¹W   …À☼DÁ‰„$►☺  ‹Èè\x90p÷ÿ\x90€¼$¨    u?è►¤öÿH…À☼„ëb  H‹Œ$°   H‰HXH\x81ÄÐ   A_A^A]A\_^[ÃÌfƒO►@ÇG¶♥   éËüÿÿ☼
O¶‹G¶‹Ðƒâþð☼±W¶uñöÂ♥u©H‹Ëèb
        ^
compilation terminated due to -fmax-errors=5.
```

### Bug Timeline

9 June 2021

### Expected behavior

Succeed build with default configuration

### Actual behavior

first error start after compiling eeprom_api.cpp.o

```
Building in release mode
Compiling .pio\build\melzi_optiboot_optimized\src\src\HAL\AVR\HAL.cpp.o
Compiling .pio\build\melzi_optiboot_optimized\src\src\HAL\AVR\HAL_SPI.cpp.o
Compiling .pio\build\melzi_optiboot_optimized\src\src\HAL\AVR\MarlinSerial.cpp.o
Compiling .pio\build\melzi_optiboot_optimized\src\src\HAL\AVR\Servo.cpp.o
Compiling .pio\build\melzi_optiboot_optimized\src\src\HAL\AVR\eeprom.cpp.o
Compiling .pio\build\melzi_optiboot_optimized\src\src\HAL\AVR\fast_pwm.cpp.o
Compiling .pio\build\melzi_optiboot_optimized\src\src\HAL\AVR\fastio.cpp.o
Compiling .pio\build\melzi_optiboot_optimized\src\src\HAL\AVR\u8g_com_HAL_AVR_sw_spi.cpp.o
Compiling .pio\build\melzi_optiboot_optimized\src\src\HAL\AVR\watchdog.cpp.o
Compiling .pio\build\melzi_optiboot_optimized\src\src\HAL\shared\Delay.cpp.o
Compiling .pio\build\melzi_optiboot_optimized\src\src\HAL\shared\HAL_MinSerial.cpp.o
Compiling .pio\build\melzi_optiboot_optimized\src\src\HAL\shared\eeprom_api.cpp.o
In file included from c:\users\diono\.platformio\packages\toolchain-atmelavr\avr\include\avr\io.h:162:0,
                 from Marlin\src\HAL\AVR\fastio.h:30,
                 from Marlin\src\HAL\AVR\fastio.cpp:29:
c:\users\diono\.platformio\packages\toolchain-atmelavr\avr\include\avr\iom1284p.h:1:1: error: stray '\215' in program
 \x8d☻  H‰†`☺  Hƒ¦h☺   A÷ ►  t◄A‹Ö3ÉèèüòÿH‰†°   H‹†ˆ   H‰†0☺  A‹☼ºà♫riL‹l$p☼ºà
☼‚\x81   H‹\$8H‹ÓH‹Îè£♂òÿ¸☻   f‰G↕H‹ÎèŠ$  ‹A+D$►A9D$↑s^¹÷♠  èar÷ÿA‹è ♦  ‹ÓH‹ÎèjèþÿH‹œ$€   é?ÿÿÿH\x8d♥  D‰l$ L‹l$pM‹ÍL‹ÃH‹Îè>ÒÿÿA‹éuÿÿÿH‹Îè²:ÿÿérÿÿÿH‹Îèõ☺  ëW‹Ø‰„$˜   ‰„$►☺  =♣  Àt§=☻  €t♫=÷♠  t=Æ♠  u♀»÷♠  ‰œ$►☺  H‹D$hfƒH►♦‰X↑H‹L$xè¶†øÿ‹Ëè­q÷ÿ\x90A÷ @  ☼…©   fD‰w↕è%¥öÿH…À☼„§   H‹HXH‰Œ$°   H‰Œ$ˆ   H‰pXDˆ´$\x90   H‹D$`H‹@ ☼·Œ$►☺  H…À☼…„   I‹D$8H…ÀH‹„$    H‹♀ÈHc‡t☻  HÁè♥‰D$@…Àt A÷  u!D‰t$4D‰t$0E☼·G↕D‹ÈI‹Õè†šùÿƒd$0 ëpA+Æ‰D$@ëÖH‹\x8fÐ☻  è’FÿÿéFÿÿÿèPCõÿH…À☼…Kÿÿÿ\x8dH♫èÏp÷ÿÌH‹♦ÈH…À☼„oÿÿÿD‰t$4D‰t$0H‹Îÿ§▬@☺ ë¦‰D$T¹W   …À☼DÁ‰„$►☺  ‹Èè\x90p÷ÿ\x90€¼$¨    u?è►¤öÿH…À☼„ëb  H‹Œ$°   H‰HXH\x81ÄÐ   A_A^A]A\_^[ÃÌfƒO►@ÇG¶♥   éËüÿÿ☼
Compiling .pio\build\melzi_optiboot_optimized\src\src\HAL\shared\esp_wifi.cpp.o
O¶‹G¶‹Ðƒâþð☼±W¶uñöÂ♥u©H‹Ëèb
 ^
```

### Steps to Reproduce

copy in Marlin-bugfix-2.0.x\Marlin folder all files from Configurations-import-2.0.x\config\examples\AnyCubic\Mega Zero 2.0\Anycubic V1
open Marlin-bugfix-2.0.x on Visual Studio Code
add PlatformIO IDE and Auto Build Marlin addons
run melzi_optiboot_optimized build

### Version of Marlin Firmware

Marlin bugfix-2.0.x

### Printer model

Anycubic Mega Zero 2.0
[... 595 lines cut ...]
☼‚\x81   H‹\$8H‹ÓH‹Îè£♂òÿ¸☻   f‰G↕H‹ÎèŠ$  ‹A+D$►A9D$↑s^¹÷♠  èar÷ÿA‹è ♦  ‹ÓH‹ÎèjèþÿH‹œ$€   é?ÿÿÿH\x8d♥  D‰l$ L‹l$pM‹ÍL‹ÃH‹Îè>ÒÿÿA‹éuÿÿÿH‹Îè²:ÿÿérÿÿÿH‹Îèõ☺  ëW‹Ø‰„$˜   ‰„$►☺  =♣  Àt§=☻  €t♫=÷♠ 
 t=Æ♠  u♀»÷♠  ‰œ$►☺  H‹D$hfƒH►♦‰X↑H‹L$xè¶†øÿ‹Ëè­q÷ÿ\x90A÷ @  ☼…©   fD‰w↕è%¥öÿH…À☼„§   H‹HXH‰Œ$°   H‰Œ$ˆ   H‰pXDˆ´$\x90   H‹D$`H‹@ ☼·Œ$►☺  H…À☼…„   I‹D$8H…ÀH‹„$    H‹♀ÈHc‡t☻  HÁè♥‰D$@…Àt A÷  u!D‰t$4D‰t$0E☼·G↕D‹ÈI‹Õè†šùÿƒd$0 ëpA+Æ‰D$@ëÖH‹\x8fÐ☻  è’FÿÿéFÿÿÿèPCõÿH…À☼…Kÿÿÿ\x8dH♫èÏp÷ÿÌH‹♦ÈH…À☼„oÿÿÿD‰t$4D‰t$0H‹Îÿ§▬@☺ ë¦‰D$T¹W   …À☼DÁ‰„$►☺  ‹Èè\x90p÷ÿ\x90€¼$¨    u?è►¤öÿH…À☼„ëb  H‹Œ$°   H‰HXH\x81ÄÐ   A_A^A]A\_^[ÃÌfƒO►@ÇG¶♥   éËüÿÿ☼
O¶‹G¶‹Ðƒâþð☼±W¶uñöÂ♥u©H‹Ëèb
   ^
c:\users\diono\.platformio\packages\toolchain-atmelavr\avr\include\avr\iom1284p.h:1:4: error: stray '\2' in program
☼‚\x81   H‹\$8H‹ÓH‹Îè£♂òÿ¸☻   f‰G↕H‹ÎèŠ$  ‹A+D$►A9D$↑s^¹÷♠  èar÷ÿA‹è ♦  ‹ÓH‹ÎèjèþÿH‹œ$€   é?ÿÿÿH\x8d♥  D‰l$ L‹l$pM‹ÍL‹ÃH‹Îè>ÒÿÿA‹éuÿÿÿH‹Îè²:ÿÿérÿÿÿH‹Îèõ☺  ëW‹Ø‰„$˜   ‰„$►☺  =♣  Àt§=☻  €t♫=÷♠ 
 t=Æ♠  u♀»÷♠  ‰œ$►☺  H‹D$hfƒH►♦‰X↑H‹L$xè¶†øÿ‹Ëè­q÷ÿ\x90A÷ @  ☼…©   fD‰w↕è%¥öÿH…À☼„§   H‹HXH‰Œ$°   H‰Œ$ˆ   H‰pXDˆ´$\x90   H‹D$`H‹@ ☼·Œ$►☺  H…À☼…„   I‹D$8H…ÀH‹„$    H‹♀ÈHc‡t☻  HÁè♥‰D$@…Àt A÷  u!D‰t$4D‰t$0E☼·G↕D‹ÈI‹Õè†šùÿƒd$0 ëpA+Æ‰D$@ëÖH‹\x8fÐ☻  è’FÿÿéFÿÿÿèPCõÿH…À☼…Kÿÿÿ\x8dH♫èÏp÷ÿÌH‹♦ÈH…À☼„oÿÿÿD‰t$4D‰t$0H‹Îÿ§▬@☺ ë¦‰D$T¹W   …À☼DÁ‰„$►☺  ‹Èè\x90p÷ÿ\x90€¼$¨    u?è►¤öÿH…À☼„ëb  H‹Œ$°   H‰HXH\x81ÄÐ   A_A^A]A\_^[ÃÌfƒO►@ÇG¶♥   éËüÿÿ☼
O¶‹G¶‹Ðƒâþð☼±W¶uñöÂ♥u©H‹Ëèb
    ^
c:\users\diono\.platformio\packages\toolchain-atmelavr\avr\include\avr\iom1284p.h:1:8: error: stray '\211' in program
 \x8d☻  H‰†`☺  Hƒ¦h☺   A÷ ►  t◄A‹Ö3ÉèèüòÿH‰†°   H‹†ˆ   H‰†0☺  A‹☼ºà♫riL‹l$p☼ºà
☼‚\x81   H‹\$8H‹ÓH‹Îè£♂òÿ¸☻   f‰G↕H‹ÎèŠ$  ‹A+D$►A9D$↑s^¹÷♠  èar÷ÿA‹è ♦  ‹ÓH‹ÎèjèþÿH‹œ$€   é?ÿÿÿH\x8d♥  D‰l$ L‹l$pM‹ÍL‹ÃH‹Îè>ÒÿÿA‹éuÿÿÿH‹Îè²:ÿÿérÿÿÿH‹Îèõ☺  ëW‹Ø‰„$˜   ‰„$►☺  =♣  Àt§=☻  €t♫=÷♠ 
 t=Æ♠  u♀»÷♠  ‰œ$►☺  H‹D$hfƒH►♦‰X↑H‹L$xè¶†øÿ‹Ëè­q÷ÿ\x90A÷ @  ☼…©   fD‰w↕è%¥öÿH…À☼„§   H‹HXH‰Œ$°   H‰Œ$ˆ   H‰pXDˆ´$\x90   H‹D$`H‹@ ☼·Œ$►☺  H…À☼…„   I‹D$8H…ÀH‹„$    H‹♀ÈHc‡t☻  HÁè♥‰D$@…Àt A÷  u!D‰t$4D‰t$0E☼·G↕D‹ÈI‹Õè†šùÿƒd$0 ëpA+Æ‰D$@ëÖH‹\x8fÐ☻  è’FÿÿéFÿÿÿèPCõÿH…À☼…Kÿÿÿ\x8dH♫èÏp÷ÿÌH‹♦ÈH…À☼„oÿÿÿD‰t$4D‰t$0H‹Îÿ§▬@☺ ë¦‰D$T¹W   …À☼DÁ‰„$►☺  ‹Èè\x90p÷ÿ\x90€¼$¨    u?è►¤öÿH…À☼„ëb  H‹Œ$°   H‰HXH\x81ÄÐ   A_A^A]A\_^[ÃÌfƒO►@ÇG¶♥   éËüÿÿ☼
O¶‹G¶‹Ðƒâþð☼±W¶uñöÂ♥u©H‹Ëèb
        ^
compilation terminated due to -fmax-errors=5.
*** [.pio\build\melzi_optiboot_optimized\src\src\MarlinCore.cpp.o] Error 1
================================================================================= [FAILED] Took 5.82 seconds ================================================================================= 

Environment               Status    Duration
------------------------  --------  ------------
melzi_optiboot_optimized  FAILED    00:00:05.825
============================================================================ 1 failed, 0 succeeded in 00:00:05.825 ============================================================================PS D:\Diono\impression 3D\Marlin-bugfix-2.0.x> echo "done" >
```
[image]

ellensp, 2021-06-09:
"copy in Marlin-bugfix-2.0.x\Marlin folder all files from Configurations-import-2.0.x\config\examples\AnyCubic\Mega Zero 2.0\Anycubic V1"
you do not ever use import branch
use bugfix for bugfix code or release for release versions of code

From the readme.md file 
```
Branches
Release: For most Marlin users. Select the version that matches the Marlin version you are compiling.

Bugfix: For users compiling the bugfix branch of Marlin.

Import: For developers maintaining the Configuration examples.
```

ellensp, 2021-06-09:
It looks like your platformio is corrupt, or perhaps your computer is full of malware. Or you hard drive is corrupted.
Download malwarebytes (in trial mode) and scan your computer (you can remove it after its done)

Run CHKDSK on your drives (google for how)

Then delete folder `c:\users\diono\.platformio`  and restart vscode, it will reinstall platformio components

Diono, 2021-06-09:
thanks a lot! yes it was corrupt files!
i delete .patformio and restart install of platformIO and now its work!

closed 2021-06-09 by Diono