MarlinFirmware/Marlin #22090: [FR] ESP32 bluetooth serial port

lols, 2021-06-09:
### Is your feature request related to a problem? Please describe.

Needs a way to connect to the esp32 via bluetooth serial port. i can make marlin to start the bluetooth and able to connect to it but not sure how to get the usb serial port and the bluetooth serial port link together 

### Are you looking for hardware support?

Using FYSETC E4 board

### Describe the feature you want

It should work like connecting a bluetooth module to a normal 32bit board so that i can send gcode via bluetooth serial

lols, 2021-06-21:
Can someone point me to a good location to include the BluetoothSerial.h? I'm not a pro coder just a noob, but i tried to include in hal.h but had a compile error with esp3dlib, end up added in in esp3dlib and slightly modified some files and surprisingly bt send and received gcode worked

thinkyhead, 2021-07-30:
I have no idea. You can use the Marlin Discord to talk to developers and get help.

ellensp, 2026-10-24:
#28747 adds this. The ESP32's built-in Bluetooth (Classic SPP) becomes a Marlin serial port, and you set it like any other serial port, with the value `-2`.

For the FYSETC E4, keep USB as the first port and add Bluetooth as the second:

```cpp
#define SERIAL_PORT 0      // USB
#define SERIAL_PORT_2 -2   // Bluetooth
```

Marlin then takes G-code from both at the same time, and answers each command on the port it came from. The board shows up under your machine name, and no pairing PIN is needed. `SERIAL_PORT` and `SERIAL_PORT_3` also accept `-2`.

Be aware that Bluetooth is a resource hog. It adds about 650 KB of flash and takes about 160 KB of RAM. That leaves enough for Marlin on its own, but not for WiFi as well, so it can't be combined with `WIFISUPPORT` or `ESP3D_WIFISUPPORT`.

closed 2026-11-04 by ellensp