MarlinFirmware/Marlin #22095: Need to show more digits. I have a bed of more than 1000 mm and it only shows up to 999.

mmatus1112, 2021-06-09:
### Describe the feature you want

I have a bed of 2000 mm by 1500 mm dimensions. I would like to see if there is any way to show 4 digits before the dot (.), lets say "Move X: +1595.1 mm". 

Thanks!!!!!!

ellensp, 2021-06-10:
What sort of display are you using? a 128x64 or a modern TFT?

mmatus1112, 2021-06-10:
Hi, yes. I am using an LCD 168x64

ellensp, 2021-06-14:
The main status screen already shows 4 digits and no decimal places. So your not talking about the status screen.

Must be talking about the move menu.

this should do it.

```DIFF
diff --git a/Marlin/src/lcd/menu/menu_motion.cpp b/Marlin/src/lcd/menu/menu_motion.cpp
index 076ece33b0..cd96b6b8f4 100644
--- a/Marlin/src/lcd/menu/menu_motion.cpp
+++ b/Marlin/src/lcd/menu/menu_motion.cpp
@@ -85,7 +85,12 @@ static void _lcd_move_xyz(PGM_P const name, const AxisEnum axis) {
       MenuEditItemBase::draw_edit_screen(name, ftostr63(imp_pos));
     }
     else
-      MenuEditItemBase::draw_edit_screen(name, ui.manual_move.menu_scale >= 0.1f ? ftostr41sign(pos) : ftostr63(pos));
+      #if X_BED_SIZE > 999 || Y_BED_SIZE > 999
+        #define MOVE_AXIS_SIGNIFICANT_FIGURES(n) ftostr51sign(n)
+      #else
+        #define MOVE_AXIS_SIGNIFICANT_FIGURES(n) ftostr41sign(n)
+      #endif
+      MenuEditItemBase::draw_edit_screen(name, ui.manual_move.menu_scale >= 0.1f ? MOVE_AXIS_SIGNIFICANT_FIGURES(pos) : ftostr63(pos));
   }
 }
 void lcd_move_x() { _lcd_move_xyz(GET_TEXT(MSG_MOVE_X), X_AXIS); }
```

ellensp, 2021-06-14:
@thinkyhead should we add this properly? more and more machines that are > 1000mm

mmatus1112, 2021-06-14:
Yes, that is correct. This is for the "Move Menu." Also, since machines are getting bigger, it would be awesome to add the option to move by 0.1 mm, 1 mm, 10 mm, and 100 mm!! Now, where do I add the code you just provided? I am using the Marlin 2.0 and I see the Config and the config_adv.

Thank you so much!

ellensp, 2021-06-14:
you have to edit the code in file Marlin/src/lcd/menu/menu_motion.cpp

ellensp, 2021-06-14:
To add a 100mm move you need this.

```DIFF
diff --git a/Marlin/src/lcd/language/language_en.h b/Marlin/src/lcd/language/language_en.h
index 33f11c9ffd..b46b6adc2a 100644
--- a/Marlin/src/lcd/language/language_en.h
+++ b/Marlin/src/lcd/language/language_en.h
@@ -281,6 +281,7 @@ namespace Language_en {
   PROGMEM Language_Str MSG_MOVE_01MM                       = _UxGT("Move 0.1mm");
   PROGMEM Language_Str MSG_MOVE_1MM                        = _UxGT("Move 1mm");
   PROGMEM Language_Str MSG_MOVE_10MM                       = _UxGT("Move 10mm");
+  PROGMEM Language_Str MSG_MOVE_100MM                      = _UxGT("Move 100mm");
   PROGMEM Language_Str MSG_MOVE_0001IN                     = _UxGT("Move 0.001in");
   PROGMEM Language_Str MSG_MOVE_001IN                      = _UxGT("Move 0.01in");
   PROGMEM Language_Str MSG_MOVE_01IN                       = _UxGT("Move 0.1in");
diff --git a/Marlin/src/lcd/menu/menu_motion.cpp b/Marlin/src/lcd/menu/menu_motion.cpp
index 076ece33b0..e344f11609 100644
--- a/Marlin/src/lcd/menu/menu_motion.cpp
+++ b/Marlin/src/lcd/menu/menu_motion.cpp
@@ -85,7 +85,12 @@ static void _lcd_move_xyz(PGM_P const name, const AxisEnum axis) {
       MenuEditItemBase::draw_edit_screen(name, ftostr63(imp_pos));
     }
     else
-      MenuEditItemBase::draw_edit_screen(name, ui.manual_move.menu_scale >= 0.1f ? ftostr41sign(pos) : ftostr63(pos));
+      #if X_BED_SIZE > 999 || Y_BED_SIZE > 999
+        #define MOVE_AXIS_SIGNIFICANT_FIGURES(n) ftostr51sign(n)
+      #else
+        #define MOVE_AXIS_SIGNIFICANT_FIGURES(n) ftostr41sign(n)
+      #endif
+      MenuEditItemBase::draw_edit_screen(name, ui.manual_move.menu_scale >= 0.1f ? MOVE_AXIS_SIGNIFICANT_FIGURES(pos) : ftostr63(pos));
   }
 }
 void lcd_move_x() { _lcd_move_xyz(GET_TEXT(MSG_MOVE_X), X_AXIS); }
@@ -170,9 +175,12 @@ void _menu_move_distance(const AxisEnum axis, const screenFunc_t func, const int
     SUBMENU(MSG_MOVE_0001IN, []{ _goto_manual_move(IN_TO_MM(0.001f)); });
   }
   else {
-    SUBMENU(MSG_MOVE_10MM, []{ _goto_manual_move(10);    });
-    SUBMENU(MSG_MOVE_1MM,  []{ _goto_manual_move( 1);    });
-    SUBMENU(MSG_MOVE_01MM, []{ _goto_manual_move( 0.1f); });
+    #if X_BED_SIZE > 999 || Y_BED_SIZE > 999
+      SUBMENU(MSG_MOVE_100MM, []{ _goto_manual_move(100);   });
+    #endif
+    SUBMENU(MSG_MOVE_10MM,  []{ _goto_manual_move( 10);    });
+    SUBMENU(MSG_MOVE_1MM,   []{ _goto_manual_move(  1);    });
+    SUBMENU(MSG_MOVE_01MM,  []{ _goto_manual_move(  0.1f); });
     if (axis == Z_AXIS && (FINE_MANUAL_MOVE) > 0.0f && (FINE_MANUAL_MOVE) < 0.1f) {
       // Determine digits needed right of decimal
       constexpr uint8_t digs = !UNEAR_ZERO((FINE_MANUAL_MOVE) * 1000 - int((FINE_MANUAL_MOVE) * 1000)) ? 4 :
```

You have to edit the language file to add the string eg Marlin/src/lcd/language/language_en.h
and add more to Marlin/src/lcd/menu/menu_motion.cpp

Vertabreak, 2021-06-14:
i would have to support changing the max to 9999 to accommodate large format printers.
as well as the addition of a 100mm move option possibly only triggered when bed_size greater then 999.
i am personally building a printer with a 2 foot by 2 foot bed.

mmatus1112, 2021-06-14:
Ellen, you are AWESOME!!!!!!!!!! Worked like a charm!!!!!! I wish I could share my project with you!!!! Thanks soooooo muchhhh!!!!!!!!!!!!!!!!!!!!!!!!!!

ellensp, 2021-06-19:
This has been added to Marlin. If your bed is >= 1000 in x or y, both the larger moves and extra digits is enabled.

closed 2021-06-19 by ellensp

Vertabreak, 2021-06-20:
what about z?

ellensp, 2021-06-21:
@Vertabreak don't be greedy!  
The  Z moves are also extended. but yes the test for a large machine should include Z axis