MarlinFirmware/Marlin #22097: [BUG] (G0 and G1 repently acts as Z numer was relative instead of absolute position)

arielito-ear, 2021-06-10:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

I have not been able to identify well when it happens, but suddenly when sending G0 or G1 commands, it interprets the position in Z as relative. For example, if I am at Z = 100 and I execute G1 Z1, instead of going to Z = 1 it goes to Z = 101, I execute G0 Z10, instead of going to Z = 10 it makes Z = Z + 1. This happens suddenly. It happened to me several times, I don't remember if they were all while trying to calibrate the bed with G33, G29, G49, M421 commands. I do not remember either if every time I spend using the repeater host, or if I also spend moving from the LCD. It is a difficult error to reproduce, since I have not yet found how to do it. If I turn off the printer and turn it back on it works fine, until at some point the problem recurs.

### Bug Timeline

2.0.8.2

### Expected behavior

1) Maintains the absolute position movement
2) Go to the absolute position.

### Actual behavior

1) Suddenly the printer change its behavior and interpret G0 Znnn and G1 Znnn as relative position.

### Steps to Reproduce

I dont, know yet how reproduce this behavior. But i think may be the bug was introducced in 2.0.8.2 with the new option to G0 and G1 command for laser. (But I'm not sure about that, I have many doubts, version 2.0.8.2 seems to be full of bugs)

### Version of Marlin Firmware

2.0.8.2

### Printer model

Generic Delta

### Electronics

SKR 1.3, TMC2209, Dual Nozzle Extruder, 1 sensor filament attached to the first extruder, TMC 2209, TFT35 (Al BigTreeTech)

### Your Slicer

Cura

### Host Software

Repetier Host

### Additional information & file uploads

[Config.zip](https://github.com/MarlinFirmware/Marlin/files/6627631/Config.zip)

arielito-ear, 2021-06-10:
Hi, i can reproduce the bug, but its i don't know how to get log in this case:
1) Strat the printer.
2) (With Repetier Host), G28
3) (With Repetier Host), G1 Z10
4) Now, go to the display (In my case a BigTreeTech TFT35 V3.0), select "move" option, and do some move in X and Y axis.
5) (With Repetier Host), G28
6) (With Repetier Host), G1 Z10
7) Now instead of move to Z=10 (Absolute move), the command do Z=Z+10 (Relative move), 

If i run M111 S247, its imposible to get information form the display, because in the first touch, starts to display a lot of messages, many of them are G90, (Thats its OK because G90 set the absolute positioning), so i don´t know how to get debug info from the display.

DerAndere1, 2021-06-11:
Thanks for the report. To exclude an error in G28, can you test if the following work-around results in absolute movement at step 7?

1. Start the printer.
2. (With Repetier Host), G28
3. (With Repetier Host), G1 Z10
4. Now, go to the display (In my case a BigTreeTech TFT35 V3.0), select "move" option, and do some move in X and Y axis.
5.  **(With Repetier Host), G90**
6.  (With Repetier Host), G28
7.  (With Repetier Host), G1 Z10

arielito-ear, 2021-06-14:
Hi, a new problem...  
**When a do this:**
1. Start the printer.
2. (With Repetier Host), G28
3. (With Repetier Host), G1 Z10
4. Now, go to the display (In my case a BigTreeTech TFT35 V3.0), select "move" option, and do some move in X and Y axis.
5. (With Repetier Host), G90 
6. (With Repetier Host), G28
7. (With Repetier Host), G1 Z10
**Result:** Z=10 (Its OK), but and if i go to the display and do some movements in X or Y, its only accepts press each buttons only one time in the same direction, is as it has limited.
Example:

Once i have reach te step 7 (Z=10), i go to the display (the distance configures in the displays shows 1mm for axis movements), and I pess +X , the nozzle move +1mm in X, if i press again +X, nothing happends, ir i press -X moves -1mm, if ia press -X again nothing happends, but if I press +X, the move +1mm, but never gos beyond of the -1mm and +1mm in X. The same with the Y axis. At first its look as was an  square on (-1,1) to (+1,1). But if i change the distance applies by the screen to 10mm, so the square has its bounds in (-10,-10) and (+10,+10), if a change agan the distance to 0.1mm the bounds change to (-0.1, -0.1) to (+0.1,+0.1). 
If move the nozzcle with repetier host, its moves OK. I aftes thast i return to the display, the limitation to movements go on.

I played some time, and after touch the display, also affects the Z axis.

Apparently it you move the nozzle using the display, it affects the correct functioning of the movement commands (I don't know if other commands are also affected)

DerAndere1, 2021-06-16:
That means the move command from the display should work as follows: 
1. save last mode (relative or absolute)
2. set to relative mode
3. move axes
4. restore last mode

Currently it seems to skip steps 1 and 4

thinkyhead, 2021-07-30:
@DerAndere1 — What are you up to? Movement from the LCD doesn't use G-code so there is no changing to relative mode. It simply goes to the new position.

thinkyhead, 2021-07-30:
@arielito-ear — Please try a different host software, or just a dumb serial terminal. Repetier Host is known to try and be too clever sometimes.

thinkyhead, 2021-07-30:
Please test the [`bugfix-2.0.x` branch](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip) to see where it stands. If the problem has been resolved then we can close this issue. If the issue isn't resolved yet, then we should investigate further.

closed 2022-05-17 by github-actions[bot]