MarlinFirmware/Marlin #22098: [BUG] Home Offset Y value does not update properly

sarvenn, 2021-06-10:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

Hi,
In TFT mode, I enter to machine setting screen and update my home offset values as X: 5 , Y:-5. Then save the settings to eeprom, then I go to terminal in TFT mode and send M503 to read back the eeprom values.

I noticed two issues.
1)In home offset screen, I enter my values above. X value shows correctly but Y value always turns back to 0. But, in the backround it is OK. I did save to eeprom command and do a homing. I saw X:5 Y:-5 Z:0 on main screen.
2) While I was sure that my Y value is save correctly but showed as 0 in homeoffset screen, I tried to readback eeprom value sending m503 from tft terminal screen.
It responded as "M206 X5 Y". That's all. I'm sure that it was correct on the older terminal view layout. May be it was also correct after new terminal layout. But I'm sure that it was broken. I updated my TFT and Marlin bugfix2 firmwares last Sunday (6th of June). My Marlin configs are also inline with the instructions for TFT.

Printer: Ender 3 Pro
Board: BTT SKR Mini E3 V2
Screen: BTT TFT 35 E3 V3
Marlin FW: Bugfix 2 (last updated on 6th of June)
TFT FW: Last updated on 6th of June

You may find the screenshot in order:
[home.offset.issue.pdf](https://github.com/MarlinFirmware/Marlin/files/6631621/home.offset.issue.pdf)

First I raised the issue at BTT TFT repo but I was suggested to raise it here since Marlin does not respond to M503 properly and therefore TFT FW can not take the correct value for Y, so it shows Z. Besides that as you can see it the pdf, once it doesn't show Z line as well, because in the response there is not even Z string, then it shows once and dissappears again.
Marlin responds to M206 if sent seperately.
Besides that correct values are save to eeprom because after homing I see correct values on the main status screen.

### Bug Timeline

It wasn't problematic in the past. After updating on 6th of June, problem started.

### Expected behavior

Proper response to M503 on M206 line.

### Actual behavior

Y value doesn't show, Z string is also missing.

### Steps to Reproduce

1. Go to machine setting, parameters, home offset menu.
2. Z line is missing. 
3. Enter 5 for X and -5 for Y using the keypad and click on tick mark.
4. Values seems to be entered correctly.
5. Go back and select save settings
6. Go to home offset again, Y value is missing.
7. Go to terminal and send m503
8. M206 respond is incorrect.
9. Send M206 respond is correct.

### Version of Marlin Firmware

6th of June version bugfix 2

### Printer model

Ender 3 pro

### Electronics

BTT SKR Mini E3 V2 and TFT35 E3 V3

ellensp, 2021-06-10:
That board has several build environment to choose from, which did you try?
env:STM32F103RC_btt 
env:STM32F103RC_btt_512K 
env:STM32F103RC_btt_maple 
env:STM32F103RC_btt_512K_maple 
env:STM32F103RC_btt_USB_maple 
env:STM32F103RC_btt_512K_USB_maple

maple is old and being retired, you shouldn't use it.
512K options are a nasty hack to use flash memory that may not exist or is faulty, Test without this option
the 512K options should only be used if main chip says STM32F103RE not RC

radek8, 2021-06-10:
I updated to the latest Marlin with bug fixes.
I have the same problem

Report M503 incorrectly reports the value of M206
Only the value of X is reported. Y is without value and Z is completely missing.

[image]

SKR V1.4 turbo
Marlin  6d2c7aa227

ellensp, 2021-06-10:
Both of you please attach your configuration files.

radek8, 2021-06-11:
the build of 5/21/2021 did not have this problem.

[Configuration.h.txt](https://github.com/MarlinFirmware/Marlin/files/6635636/Configuration.h.txt)
[Configuration_adv.h.txt](https://github.com/MarlinFirmware/Marlin/files/6635637/Configuration_adv.h.txt)

sarvenn, 2021-06-11:
I am away from home now so I cant attach my config files now, but just like radek, I didnt have this issue at an older build on May.
For a year I use 512k environment without problem.i but I will check my board and read the chip code, I hope it is written on it. I didnt use maple.

DerAndere1, 2021-06-11:
Please connect your printer via USB and send  G-code command M504 using a host software such as Printrun or Repetier Host. The host software has a terminal/serial monitor where messages from the printer are logged. What is the response from the printer?  

If an error message is reported in the host software, please enable DEBUG_EEPROM_READWRITE by changing settings.cpp line 44 from

https://github.com/MarlinFirmware/Marlin/blob/1daee11db9eca45dfe0cde2be5fdf537224b2af7/Marlin/src/module/settings.cpp#L44

to 
```
#define DEBUG_EEPROM_READWRITE
```
Then flash the Firmware again, send M502, then M500. Finally send M504 and report the response from the printer here

Probably caused by https://github.com/MarlinFirmware/Marlin/pull/21953 or https://github.com/MarlinFirmware/Marlin/pull/19112 .

radek8, 2021-06-11:
[image]

[image]

radek8, 2021-06-11:
The values are OK only the message from M503 is corrupted

DerAndere1, 2021-06-11:
Would be great if you can test changing settings.cpp line 3166

https://github.com/MarlinFirmware/Marlin/blob/1daee11db9eca45dfe0cde2be5fdf537224b2af7/Marlin/src/module/settings.cpp#L3166

to the following:
```
          LIST_N(DOUBLE(LINEAR_AXES),
```

radek8, 2021-06-11:
It works great
Thank you

[image]
[image]

thisiskeithb, 2021-06-11:
#22112 was merged. Closing.

closed 2021-06-11 by thisiskeithb

Buzzz57, 2021-08-10:
I still get this issue with the latest 2.0.x build.
My settings.cpp file has the "LIST_N(DOUBLE(LINEAR_AXES),"
Where should I look ?