MarlinFirmware/Marlin #22101: [BUG] Unexpected error from #22086 #define Z_MIN_PROBE_ENDSTOP_INVERTING false

bill-orange, 2021-06-10:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

 #error "BLTOUCH requires Z_MIN_ENDSTOP_INVERTING set to false. Please update your Configuration.h file."  error thrown even when `#define Z_MIN_PROBE_ENDSTOP_INVERTING false  // Set to true to invert the logic of the probe.`

Can not compile.

### Bug Timeline

Latest release only.  Appears to be from #22086

### Expected behavior

Normal compile with hoded define set to false

### Actual behavior

Throws error and will not compile

### Steps to Reproduce

1. Be sure #define Z_MIN_PROBE_ENDSTOP_INVERTING false and try to compile
2. Note error

### Version of Marlin Firmware

Latest Bugfix

### Printer model

Anet A8 with SKR1.3

### Electronics

SKR1.3

### Add-ons

BLTouch (not a clone)

### Your Slicer

Cura

### Host Software

OctoPrint

### Additional information & file uploads

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/6634226/Marlin.zip)

ellensp, 2021-06-10:
The error is error: #error "BLTOUCH requires Z_MIN_ENDSTOP_INVERTING set to false. Please update your Configuration.h file."

you have #define Z_MIN_ENDSTOP_INVERTING true in your Configuration.h

closed 2021-06-10 by ellensp

bill-orange, 2021-06-10:
@ellensp I should have communicated more clearly.  It doesn't matter if it is set true or false, You get the same error regardless.  Please reopen.

ellensp, 2021-06-10:
Please read the rrro 
this is Z_MIN_ENDSTOP_INVERTING not Z_MIN_PROBE_ENDSTOP_INVERTING

bill-orange, 2021-06-10:
@ellensp Got it, I can compile now.  However with the offending selections in the define statements, my BLTouch worked fine.  What's going on here?  Will it still work with these #define changes?

ellensp, 2021-06-10:
You looked  at the PR "remove bltouch auto settings, use error instead of warnings" , 
Originally Marlin reset these setting to what was required completely unseen by the user. Causing much confusion as to what the real requirements where, and numerous posts that where just wrong regarding the required settings.

I originally added warnings that marlin was resetting some config settings https://github.com/MarlinFirmware/Marlin/pull/22069 
But Thinkyhead insisted that the warnings be errors so the Config files are accurate, and not just made to work regardless of bad settings.

Arnie3d, 2021-08-01:
BLTOUCH requires Z_MIN_PROBE_ENDSTOP_INVERTING set to false. Please update your Configuration.h file does that mean i can not change to " True"  im using 2.0.9.1 newest Marlin Firmware 2.0.8.0 it worked but now you get this why ?

victor7376, 2021-08-03:
> BLTOUCH requires Z_MIN_PROBE_ENDSTOP_INVERTING set to false. Please update your Configuration.h file does that mean i can not change to " True" im using 2.0.9.1 newest Marlin Firmware 2.0.8.0 it worked but now you get this why ?

Would love to know myself as YouTube guides state to put the setting to 'true', but the latest firmware refuses that setting.

thisiskeithb, 2021-08-03:
> Would love to know myself as YouTube guides state to put the setting to 'true', but the latest firmware refuses that setting.

YouTube videos & older guides are outdated. Follow the error message and set it to false.

---

This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- [Marlin Documentation](https://marlinfw.org) hosted on MarlinFW.org
- [Marlin Forum](https://forums.reprap.org/list.php?415) hosted on RepRap.org
- [Marlin Firmware](https://www.facebook.com/groups/1049718498464482/) Facebook Group 
- [Marlin Firmware for 3D Printers](https://www.facebook.com/groups/3Dtechtalk/) Facebook Group
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- [Marlin Firmware](https://discord.gg/n5NJ59y) Discord server