MarlinFirmware/Marlin #22103: [BUG] G12 misses documentation for XYZ

dwery, 2021-06-11:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

G12's documentation doesn't explain what the code is doing with XYZ params
```
/**
 * G12: Clean the nozzle
 *
 *  E<bool>          : 0=Never or 1=Always apply the "software endstop" limits
 *  P0 S<strokes>    : Stroke cleaning with S strokes
 *  P1 Sn T<objects> : Zigzag cleaning with S repeats and T zigzags
 *  P2 Sn R<radius>  : Circle cleaning with S repeats and R radius
 */

```
```
  const bool seenxyz = parser.seen("XYZ");
  const uint8_t cleans =  (!seenxyz || parser.boolval('X') ? _BV(X_AXIS) : 0)
                        | (!seenxyz || parser.boolval('Y') ? _BV(Y_AXIS) : 0)
                        | TERN(NOZZLE_CLEAN_NO_Z, 0, (!seenxyz || parser.boolval('Z') ? _BV(Z_AXIS) : 0))
                      ;

```

### Version of Marlin Firmware

n/a

slowbro, 2021-06-11:
I don't think it's used anymore. Trolling through the history on the command, specifically in `libs/nozzle.cpp`, it was first used to determine the 'end' variable:
https://github.com/MarlinFirmware/Marlin/commit/50e4545255605eb506c20eb107270038b0fe7bdb#diff-d0a65d061cd90ab2729f8ba8fedf4688566a15f1a369fe66fbbfe14207c96759R167 (long load, huge commit)

it was later changedslightly: 
https://github.com/MarlinFirmware/Marlin/commit/b8cc61262f5d4f322f537e7eb2ac75bec9490249#diff-d0a65d061cd90ab2729f8ba8fedf4688566a15f1a369fe66fbbfe14207c96759R169

..and finally, the useful code in that block was removed seemingly in favor of new `xyz_pos_t middle[HOTENDS]`:
https://github.com/MarlinFirmware/Marlin/commit/79810b7fcb4149896180d757e2fbb1ec7a8fc217#diff-d0a65d061cd90ab2729f8ba8fedf4688566a15f1a369fe66fbbfe14207c96759R160

...so, I think this just slipped through the cracks over the years. I'll clean it up a bit.

thinkyhead, 2021-06-12:
Added:
```
 *  X, Y, Z          : Specify axes to move during cleaning. Default: ALL.
```

closed 2021-06-12 by thinkyhead