MarlinFirmware/Marlin #22104: Error Loading Marlin Bugfix Code for Arduino DUE

Haiang, 2021-06-11:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

When I load Marlin Bugfix code for Arduino DUE it gives error

Looking for chip.h dependency? Check our library registry!
*
* CLI > platformio lib search "header:chip.h"
* Web > https://platformio.org/lib/search?query=header:chip.h
*
************************************************** ************

  #include <chip.h>
           ^~~~~~~~
compilation terminated.
*** [.pioenvs\DUE\src\src\HAL\HAL_DUE\HAL_timers_Due.cpp.o] Error 1
*** [.pioenvs\DUE\src\src\HAL\HAL_DUE\HAL_spi_Due.cpp.o] Error 1
 
I don't know how to handle it now, please help
The platformio.ini configuration I set up is as follows
[platformio]
src_dir = Marlin
build_dir = .pioenvs
lib_dir = .piolib
libdeps_dir = .piolibdeps
env_default = DUE

### Bug Timeline

New

### Expected behavior

Don't know how to fix the error

### Actual behavior

hope the helping

### Version of Marlin Firmware

Marlin bugfix 2x

### Printer model

SLA Marlin

### Electronics

Arduino DUE

ellensp, 2021-06-11:
Please attach your configuration files.

Haiang, 2021-06-11:
> Vui lòng đính kèm tệp cấu Can I send you Gmail?hình của bạn.

Haiang, 2021-06-11:
Can I send you Gmail?
give me your gmail address i send you my files tệp

thisiskeithb, 2021-06-11:
> Can I send you Gmail?
> give me your gmail address i send you my files tệp

Attach them to this issue as a **.zip** file.

ellensp, 2021-06-18:
7 days, no configs. closing.

closed 2021-06-18 by ellensp