MarlinFirmware/Marlin #22113: [BUG] bugfix-2.0 fails to build "as is" (Configuration_adv.h related)

JeanLucCoulon, 2021-06-11:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

Hi,

I've downloaded 2-0 bugfix (commit 2116e4202b064c6cafef70d54ed50edf00b79e44) and Configuration-import-2.0.x (commit 6f007ddb1ef27e7952855dae939132e72d6a9abe).

From configuration, I use Configuration_adv.h and Configuration from ./config/examples/BIQU/B1/

From vscode, it fails to build.

I've done some modifications to Configuration_adv.h (see later) and managed to have it to build cleanly.

Regards 

Jean-Luc

### Steps to Reproduce

1 - I tried to build Marlin is with the supplied Configuration.h and Configuration_adv.h

2 - I get a build error (see the file 
[marlin-2.0-bugfix-git-build-report-failed.txt](https://github.com/MarlinFirmware/Marlin/files/6640479/marlin-2.0-bugfix-git-build-report-failed.txt)

3 - I patched Configuration_adv.h to comment out some recently added statement (#if AXIS_IS_TMC(I)  and others, see the file [Configuration_adv.patch.txt](https://github.com/MarlinFirmware/Marlin/files/6640459/Configuration_adv.patch.txt)

4 - I got again a build error related to the version of the configuration files, see the attached report ([marlin-2.0-bugfix-git-build-report-failed-2.txt](https://github.com/MarlinFirmware/Marlin/files/6640463/marlin-2.0-bugfix-git-build-report-failed-2.txt)

5 - I've then edited manually the configuration version and changed it, in both the files from 02000900 to 02000801
#define CONFIGURATION_ADV_H_VERSION 02000801
#define CONFIGURATION_H_VERSION 02000801

Then I succeed building Marlin

### Version of Marlin Firmware

2.0.8.2

### Printer model

BigTreeTech BIQU B1

DerAndere1, 2021-06-11:
Can you try again with current bugfix-2.0.x?
Please also verify that the status bar indicates in the bottom left corner that you have really checked out branch bugfix-2.0.x.

JeanLucCoulon, 2021-06-11:
I work with linux. Attached, the version I'm using. I've cloned it.
[image]

I don't know how to checkout the branch from vscode.

ManuelMcLure, 2021-06-11:
You cannot use the `import-2.0.x` branch of the configuration repository. You need to build the `bugfix-2.0.x` branch of Marlin with the configuration files from the `bugfix-2.0.x` branch of the configuration repository.

thisiskeithb, 2021-06-11:
Something with your download/build is broken. Start with fresh copies and try again.

As a test, I started with the latest `bugfix-2.0.x` (https://github.com/MarlinFirmware/Marlin/commit/1daee11db9eca45dfe0cde2be5fdf537224b2af7) and tried both Biqu B1 configs from the [bugfix-2.0.x Configurations branch](https://github.com/MarlinFirmware/Configurations/tree/bugfix-2.0.x/config/examples/BIQU), they build fine without needing to edit files.

---

This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- [Marlin Documentation](https://marlinfw.org) hosted on MarlinFW.org
- [Marlin Forum](https://forums.reprap.org/list.php?415) hosted on RepRap.org
- [Marlin Firmware](https://www.facebook.com/groups/1049718498464482/) Facebook Group 
- [Marlin Firmware for 3D Printers](https://www.facebook.com/groups/3Dtechtalk/) Facebook Group
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- [Marlin Firmware](https://discord.gg/n5NJ59y) Discord server

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2021-06-11 by thisiskeithb

DerAndere1, 2021-06-11:
@JeanLucCoulon you can follow the VSCode instructions at https://code.visualstudio.com/docs/editor/versioncontrol including the [first linked video](https://code.visualstudio.com/docs/introvideos/versioncontrol)

Alternatively you can click `PlatformIO -> Quick Access -> Miscellaneous -> Clone Git Project` and after cloning, you can change to a different branch by clicking on the current branch name in the bottom left corner of the [VS Code status bar](https://code.visualstudio.com/docs/editor/versioncontrol#_git-status-bar-actions) )

JeanLucCoulon, 2021-06-13:
On the GitHub page, the link I get from "Code" is wrong 
[image]