MarlinFirmware/Marlin #22114: Need a fix on Marlin 2.0.8.2 driver file

TomTay56, 2021-06-11:
### Did you test the latest `bugfix-2.0.x` code?

No, but I will test it now!

### Bug Description

Marlin does not build. 
There is an error in the code.

### Bug Timeline

6/11/2021

### Expected behavior

Expected a good build.

### Actual behavior

Firmware did not build.

### Steps to Reproduce

Try and build with same board. Listed in photo.

### Version of Marlin Firmware

2.8.0.2

### Printer model

Ender-3

### Electronics

SKR Mini E3 V2.0

### Add-ons

None

### Your Slicer

Cura

### Additional information & file uploads

[image]

alexandremstf, 2021-06-11:
Hi, i have a similar problem but in another file (stepper.cpp). I couldnt compile without change this  (A##_HOME_TO_MIN)  to this (A##_HOME_DIR < 0). In the code below: 

`#define DUAL_ENDSTOP_APPLY_STEP(A,V)                                                                                        \
  if (separate_multi_axis) {                                                                                                \
    if (A##_HOME_DIR < 0) {                                                                                                  \
      if (TERN0(HAS_##A##_MIN, !(TEST(endstops.state(), A##_MIN) && count_direction[_AXIS(A)] < 0) && !locked_##A##_motor)) A##_STEP_WRITE(V);     \
      if (TERN0(HAS_##A##2_MIN, !(TEST(endstops.state(), A##2_MIN) && count_direction[_AXIS(A)] < 0) && !locked_##A##2_motor)) A##2_STEP_WRITE(V); \
    }                                                                                                                       \
    else {                                                                                                                  \
      if (TERN0(HAS_##A##_MAX, !(TEST(endstops.state(), A##_MAX) && count_direction[_AXIS(A)] > 0) && !locked_##A##_motor)) A##_STEP_WRITE(V);     \
      if (TERN0(HAS_##A##2_MAX, !(TEST(endstops.state(), A##2_MAX) && count_direction[_AXIS(A)] > 0) && !locked_##A##2_motor)) A##2_STEP_WRITE(V); \
    }                                                                                                                       \
  }                                                                                                                         \
  else {                                                                                                                    \
    A##_STEP_WRITE(V);                                                                                                      \
    A##2_STEP_WRITE(V);                                                                                                     \
  }`

thinkyhead, 2021-06-11:
This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- Marlin Discord server. Join link: https://discord.gg/n5NJ59y
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- [Tom's 3D Forums](https://forum.toms3d.org/)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2021-06-11 by thinkyhead

alexandremstf, 2021-06-11:
Hi, i have a similar problem but in another file (stepper.cpp). I couldnt compile without change this (A##_HOME_TO_MIN) to this (A##_HOME_DIR < 0). In the code below:

  if (separate_multi_axis) {                                                                                                \
    if (A##_HOME_DIR < 0) {                                                                                                  \
      if (TERN0(HAS_##A##_MIN, !(TEST(endstops.state(), A##_MIN) && count_direction[_AXIS(A)] < 0) && !locked_##A##_motor)) A##_STEP_WRITE(V);     \
      if (TERN0(HAS_##A##2_MIN, !(TEST(endstops.state(), A##2_MIN) && count_direction[_AXIS(A)] < 0) && !locked_##A##2_motor)) A##2_STEP_WRITE(V); \
    }                                                                                                                       \
    else {                                                                                                                  \
      if (TERN0(HAS_##A##_MAX, !(TEST(endstops.state(), A##_MAX) && count_direction[_AXIS(A)] > 0) && !locked_##A##_motor)) A##_STEP_WRITE(V);     \
      if (TERN0(HAS_##A##2_MAX, !(TEST(endstops.state(), A##2_MAX) && count_direction[_AXIS(A)] > 0) && !locked_##A##2_motor)) A##2_STEP_WRITE(V); \
    }                                                                                                                       \
  }                                                                                                                         \
  else {                                                                                                                    \
    A##_STEP_WRITE(V);                                                                                                      \
    A##2_STEP_WRITE(V);                                                                                                     \
  }

thinkyhead, 2021-06-11:
@alexandremstf — Please don't SPAM other issues with unrelated questions. (But thanks for the info. The solution is to use `ENABLED(A##_HOME_TO_MIN)` because the symbol may not be defined.)

thinkyhead, 2021-06-11:
…and the patch is already in Marlin, so again, please check the latest code before reporting things.