MarlinFirmware/Marlin #22115: Marlin keeps looping a button press on bugfix 2.0.x ender 5 creality 32bit board

cripto101, 2021-06-12:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

When I start my ender 5, it performs a button press every second or less.  I have flashed marlin 2 different times to make sure I did not change anything, and I still get the same error.  I upgraded the board to the 4.2.7 board, and this is how it is responding.

### Bug Timeline

Just started today

### Expected behavior

Start marlin and it is ready to take input

### Actual behavior

The button is triggered on its own every second or less.

### Steps to Reproduce

No clue

### Version of Marlin Firmware

Bugfix 2.0.x

### Printer model

Creality ender 5

### Electronics

4.2.7 board

### Add-ons

swiss all metal filament runout and bltouch

### Your Slicer

Cura

### Host Software

SD Card (headless)

### Additional information & file uploads

[Marlin Cripto configs.zip](https://github.com/MarlinFirmware/Marlin/files/6641485/Marlin.Cripto.configs.zip)

cripto101, 2021-06-12:
On a side note, I did see that when the printer is powering down, the button pressing stops.  This happens when the switch is off, but the power lingers for a few seconds.

ellensp, 2021-06-12:
what happens if you power board from usb only (with mains unplugged) ?

cripto101, 2021-06-12:
I tried this, and it boots normally.  I first started it up without everything plugged in.  I just had to replace the board due to the last one popping and smoking.  I started to plugin everything one item at a time.  As soon as I started the printer will only LCD plugged in it made the beeping and button press less than every 1 second.  As I plugged in the bed and hotend, it slowed down to every 3 to 5 seconds.  Once I did the fans, it became even more infrequent.  I think it was doing it due to not having everything plugged in.

XDA-Bam, 2021-06-20:
Honestly: With your former board smoked and all, this sounds like a hardware defect to me. Maybe your display/encoder got damaged by the past magic smoke event. I would suggest looking into that first, before starting with firmware diagnostics.

arggwhatisnttaken, 2021-06-25:
I am also experiencing the same issue. My previous board was functional - printer functional prior to mb swap. I have an Ender 5 and moved to a 4.2.7 board to experience the silence but I am having the same experience of button presses on its own. I am running the most recent bugfix which I downloaded days ago.

arggwhatisnttaken, 2021-06-26:
Upon further investigation, I installed Marlin 2.0.x and Bugfix 2.0.x. I have installed Creality's version of Marlin for the Ender 5 as well as TH3d. All have given the same experience. Being that others are experiencing this issue as well, I feel this is a bug and not a hardware failure. I am not ruling hardware out but I do believe it is unlikely. 

Thank you,

XDA-Bam, 2021-06-26:
> Upon further investigation, I installed Marlin 2.0.x and Bugfix 2.0.x. I have installed Creality's version of Marlin for the Ender 5 as well as TH3d. All have given the same experience. Being that [...]

I have no stakes in this and I'm happy to help diagnose bugs and problems. And I have to say that I do not follow your logic in this case. There's multiple tenthousand 4.2.7 boards out there, with people running them on different printers with many different firmwares including the newest bugfix Marlin. There's two reports of invonluntary button presses. First argument against a software bug for me.

Second argument: You noted yourself that you tried various different firmwares from different vendors and with different compile/release times. And to my understanding, your problem persisted with all of them. So, either the identical software bug is present in all available firmwares from everybody and since a long time (which is possible, but very unlikely), or it's a hardware defect on a handful of Creality boards (seems much more likely to me, in this case). Am I missing something?

thisiskeithb, 2021-06-26:
> There's multiple tenthousand 4.2.7 boards out there, with people running them on different printers with many different firmwares including the newest bugfix Marlin.

Indeed. These issues are especially common for Ender-5 owners and one of the options below usually fixes it:

- Shield LCD cable with faraday tape
- Install an EMI filter board
- Add more grounding straps to your LCD/PSU/frame/motors/etc.

I just flashed the latest `bugfix-2.0.x` (a37cc76) to a stock Crealtiy 4.2.7 board with a genuine `CR10_STOCKDISPLAY` and there are no ghost presses.

Like @XDA-Bam mentioned above, we'd have a lot more feedback by now if there was a Marlin-wide bug with these boards/LCDs due to their popularity.

closed 2021-06-26 by thisiskeithb