MarlinFirmware/Marlin #22117: [BUG] Bugfix 02000900 Compilation errors

qystan, 2021-06-12:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

Had a series of errors while compiling under bugfix 02000900.  Compiles fine under version 02000801.  

Board is Chitu V8-190729, defined as BOARD_CHITU3D_V5.

Config files attached.

Errors list follows:

Marlin\src\lcd\tft_io\tft_io.cpp: In static member function 'static void TFT_IO::set_window(uint16_t, uint16_t, uint16_t, uint16_t)':
Marlin\src\lcd\tft_io\tft_io.cpp:189:19: error: 'ILI9341_CASET' was not declared in this scope
       io.WriteReg(ILI9341_CASET);
                   ^~~~~~~~~~~~~
Marlin\src\lcd\tft_io\tft_io.cpp:189:19: note: suggested alternative: 'ILI9488_CASET'
       io.WriteReg(ILI9341_CASET);
                   ^~~~~~~~~~~~~
                   ILI9488_CASET
Marlin\src\lcd\tft_io\tft_io.cpp:196:19: error: 'ILI9341_PASET' was not declared in this scope
       io.WriteReg(ILI9341_PASET);
                   ^~~~~~~~~~~~~
Marlin\src\lcd\tft_io\tft_io.cpp:196:19: note: suggested alternative: 'ILI9488_PASET'
       io.WriteReg(ILI9341_PASET);
                   ^~~~~~~~~~~~~
                   ILI9488_PASET
Marlin\src\lcd\tft_io\tft_io.cpp:203:19: error: 'ILI9341_RAMWR' was not declared in this scope
       io.WriteReg(ILI9341_RAMWR);
                   ^~~~~~~~~~~~~
Marlin\src\lcd\tft_io\tft_io.cpp:203:19: note: suggested alternative: 'ILI9488_RAMWR'
       io.WriteReg(ILI9341_RAMWR);
                   ^~~~~~~~~~~~~
                   ILI9488_RAMWR
Marlin\src\lcd\tft_io\tft_io.cpp:210:19: error: 'ILI9328_HASTART' was not declared in this scope
       io.WriteReg(ILI9328_HASTART);
                   ^~~~~~~~~~~~~~~
Marlin\src\lcd\tft_io\tft_io.cpp:210:19: note: suggested alternative: 'ILI9488_PASET'
       io.WriteReg(ILI9328_HASTART);
                   ^~~~~~~~~~~~~~~
                   ILI9488_PASET
Marlin\src\lcd\tft_io\tft_io.cpp:212:19: error: 'ILI9328_HAEND' was not declared in this scope
       io.WriteReg(ILI9328_HAEND);
                   ^~~~~~~~~~~~~
Marlin\src\lcd\tft_io\tft_io.cpp:212:19: note: suggested alternative: 'ILI9488_RAMRD'
       io.WriteReg(ILI9328_HAEND);
                   ^~~~~~~~~~~~~
                   ILI9488_RAMRD

### Bug Timeline

Problem only appears when compiling under bugfix 02000900.

### Version of Marlin Firmware

02000900

### Electronics

Chitu V8-190729

### Additional information & file uploads

[config.zip](https://github.com/MarlinFirmware/Marlin/files/6641695/config.zip)

ellensp, 2021-06-12:
Provided configs are broken 
  #define MOTHERBOARD BOARD_BOARD_CHITU3D_V5    //  V8-190729 is similar to the V5
  #define MOTHERBOARD BOARD_CHITU3D_V5    //  V8-190729 is similar to the V5 

BABYSTEP_MULTIPLICATOR_Z is way large

but eventually you do get this error

ellensp, 2021-06-12:
@rhapsodyv Can you take a look at this?

qystan, 2021-06-12:
Sorry for the wrong config files. 

[2.0.x config.zip](https://github.com/MarlinFirmware/Marlin/files/6642064/2.0.x.config.zip)

ellensp, 2021-06-14:
The fix for this has be added to the current bugfix. Closing

PR #22125 (2021-06-14): 🐛 Some ILI9328 / ILI9341 macros apply to all TFT drivers
### Description

Fix compilation error after #22031

### Related Issues

Fixes #22117
closed 2021-06-14 by ellensp