MarlinFirmware/Marlin #22118: [BUG] (UBL "Fine Tune All" not work)

npmomchev, 2021-06-12:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

Feature:
MENU -> Motion -> Unified Bed Leveling -> Step-By-Step UBL -> Fine Tune All
Not Work

### Bug Timeline

old bug, but I'm not sure when

### Expected behavior

Fine Tune all mesh points one by one

### Actual behavior

Nothing happens just by hearing a short beep

### Steps to Reproduce

Use "Step-By-Step UBL" Feature.

After doing it properly
step 1. step 2. and step 3.
step 4. not working

### Version of Marlin Firmware

Recv: FIRMWARE_NAME:Marlin bugfix-2.0.8.2 (Jun  4 2021 04:34:26)

### Printer model

coreXY kinematics DIY 

### Electronics

BTT SKRpro v1.1

### Your Slicer

Cura

### Host Software

OctoPrint

### Additional information & file uploads

I found a working solution for me.

In file: "ubl_G29.cpp" after replacing "G29P4R999T" with "G29P4RT" on line 597 and line 599,  
function works as expected for me

If the number of Мesh points is not explicitly set with the R parameter, UBL calculates their number.

This is a report from the console after sending the wrong command in my opinion:

Send: G29P4R999T
Recv: //action:notification Doing G29
Recv: ?(R)epetition count invalid (1+).
Recv: 
Recv: ok

slowbro, 2021-06-13:
I think it's because `R_repetiton`, the variable used to hold the repetiton count, is an `int8_t` - i.e. signed 8 big integer. '999' overflows a bunch and finally ends up at '-25'. Previously, it was a `static int` which would hold '999' without issue; looks like this was changed in da4b6896f7e4f102d8c2164e7aecf22cf2922fe2.

I'll fix up the menu items, where it still has `R999` - you were correct on that solution.

slowbro, 2021-06-14:
#22129 is merged; this should be ready to close.

ellensp, 2021-06-14:
closing. if issue persists we can reopen.

closed 2021-06-14 by ellensp

PatrickNTPC, 2021-07-15:
I updated Marlin to 2.0.9.1. But I still have the same problem. How can I fix it?

slowbro, 2021-07-16:
Can you try resetting your EEPROM? M502 to factory reset followed by M500 to  save to EEPROM. Perhaps that could be causing issues. Can you also upload your configs?

PatrickNTPC, 2021-07-16:
@slowbro 
When I update the firmware, I have initialized EEROM and factory reset. This is my configs, maybe I have made some stupid settings . Thank you for your help. 
[marlin_2091_config.zip](https://github.com/MarlinFirmware/Marlin/files/6827001/marlin_2091_config.zip)