MarlinFirmware/Marlin #22119: [BUG] BFT error after commit 08155b4875fdb08f1888cacae6e1b68b48b9e51f

mriscoc, 2021-06-12:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

Binary File Transfer add a 00 00 to the start of the files and then replace 00 00 each 512 bytes to the file saved on the SD card.
That behavior is present only after apply the commit https://github.com/MarlinFirmware/Marlin/commit/08155b4875fdb08f1888cacae6e1b68b48b9e51f

### Version of Marlin Firmware

Latest bugfix 2.0.x

### Printer model

Creality Ender 3v2

### Electronics

STM32F103RET6

### Host Software

OctoPrint

### Additional information & file uploads

With: python ./transfer.py TEST.GCO -b 250000 -p /dev/ttyUSB0 -d 50 -c
There is an error in the saved file.

With: python ./transfer.py TEST.GCO -b 250000 -p /dev/ttyUSB0 -d 50
The file is saved ok.

Then, there is an error in the compression buffer.

mriscoc, 2021-06-14:
Environment STM32F103RET6_creality fails to define ARDUINO_ARCH_STM32F1 that is necessary for the binary_stream.h conditional compile:
#if BOTH(ARDUINO_ARCH_STM32F1, SDIO_SUPPORT)

ellensp, 2021-06-14:
Confirmed.
I just changed the #if BOTH(ARDUINO_ARCH_STM32F1, SDIO_SUPPORT) to #if 1 
And tested. Compressed uploads are no longer corrupt

ellensp, 2021-06-14:
#if  EITHER(ARDUINO_ARCH_STM32F1, STM32F1) && ENABLED(SDIO_SUPPORT)

should fix it, just testing.

ellensp, 2021-06-15:
A fix has been merged. closing

closed 2021-06-15 by ellensp