MarlinFirmware/Marlin #22120: Power Loss Recovery (PLR) performs Linear NOZZLE_CLEAN

qystan, 2021-06-12:
### Did you test the latest `bugfix-2.0.x` code?

No, but I will test it now!

### Bug Description

This is running under version 02000801 with NOZZLE_CLEAN_FEATURE enabled.  I am not able to test it out under bugfix due to compilation errors (see Issue #22117).

The PLR recovery steps observed:

1. Nozzle and bed reheats.
2. Performs an Auto-home.
3. Performs a linear nozzle clean if NOZZLE_CLEAN_FEATURE is active at nozzle (Z_MIN_POS+1).  Does not occur is NOZZLE_CLEAN_FEATURE is not activated.
4. Continues with object printing.

The problem is with step 3.  With PLR, an object is on the bed and will be knocked off by the hot end/housing if it is in the vicinity of the nozzle clean strokes.  I have modified the nozzle clean movement to 0 on the X-axis to minimise its intrusion into the bed area and deactivate PLR if the print goes into the cleaning zone

 #define NOZZLE_CLEAN_START_POINT { {  0, 30, (Z_MIN_POS + 1) } }
 #define NOZZLE_CLEAN_END_POINT   { { 0, 60, (Z_MIN_POS + 1) } }

A safer implementation will be to pause the hotend after homing for the user to perform an manual clear and continue with user.

Also consider a user caution to keep the hotend housing envelope at home location clear when PLR is enabled as there will be homing actions with partial models on the bed.  Could be as an added comment in the PLR definition section.

### Bug Timeline

Unknown.  Only noticed when PLR was triggered recently.

### Expected behavior

PLR should not perform nozzle cleaning during restart.

### Steps to Reproduce

1. Start print.  Print must physically commence for PLR to work.
2. Simulate power failure by switching off supply.
3. Reinstate power and observe.

### Version of Marlin Firmware

2.0 02000801

### Printer model

Tronxy XY3

### Electronics

Chitu V8-190729

### Your Slicer

Cura

### Additional information & file uploads

[2.0 config.zip](https://github.com/MarlinFirmware/Marlin/files/6641780/2.0.config.zip)

ellensp, 2021-06-14:
Results on testing on bugfix?

qystan, 2021-06-14:
Tested on bugfix, same.  Updated issue comment to reflect.  Thanks.

thinkyhead, 2021-07-30:
Introduced in #18558 by @sjasonsmith, probably because there is a nozzle purge just before the clean. I suppose we can make this optional, so it is not a bug but a feature request.