MarlinFirmware/Marlin #22126: [BUG] G28 documentation does not explain L parameter

dwery, 2021-06-12:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

With bed leveling enabled, G28 has an additional L parameter that's not explained in the docs.

### Version of Marlin Firmware

n/a

Roxy-3D, 2021-06-14:
Incidentally, for anybody wondering....   The 'L' option disables the Bed Leveling matrix prior to homing the extruder.

  // Disable the leveling matrix before homing
  #if HAS_LEVELING
    const bool leveling_restore_state = parser.boolval('L', TERN(RESTORE_LEVELING_AFTER_G28, planner.leveling_active, ENABLED(ENABLE_LEVELING_AFTER_G28)));
    IF_ENABLED(PROBE_MANUALLY, g29_in_progress = false); // Cancel the active G29 session
    set_bed_leveling_enabled(false);
  #endif

dwery, 2021-06-15:
Uhm, it looks more like it’s handling the restoring of the previous state

slowbro, 2021-06-16:
In bugfix-2.0.x, it looks like this:

```cpp
  // Disable the leveling matrix before homing
  #if CAN_SET_LEVELING_AFTER_G28
    const bool leveling_restore_state = parser.boolval('L', TERN1(RESTORE_LEVELING_AFTER_G28, planner.leveling_active));
  #endif

// ... skip a bunch of lines ...

  TERN_(CAN_SET_LEVELING_AFTER_G28, if (leveling_restore_state) set_bed_leveling_enabled());                                                                                                                                                   

```

So, it's controlling if the leveling should be restored after homing. The default is yes, and it will return to whatever the state is (`parser.leveling_active`). If you supply `L0` it will not restore the leveling state (but will still diable it before homing).

slowbro, 2021-06-19:
Documentation merged, this can be closed.

closed 2021-06-19 by thisiskeithb