MarlinFirmware/Marlin #22139: [BUG/DOCS] G28Z with MULTIPLE_PROBING doesn't behave as expected without HOME_BUMP_MM

slowbro, 2021-06-15:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

Beginning a while back, I noticed that my Z probe no longer double-checked the bed, but was not able to relate it to any changes made at the time (since I had been making a lot of changes/updated/etc). I rolled with it for a while, but got curious, and believe I found the answer. Around that time, I switched to `SENSORLESS_HOMING` for X and Y. In the comment above it, it says:

```
  /**
   * Use StallGuard to home / probe X, Y, Z.
   *
   ...
   * It is recommended to set HOMING_BUMP_MM to { 0, 0, 0 }.
```

I did as instructed, even though I am not using sensorless homing for Z. As it turns out, this seems to be what caused the `MULTIPLE_PROBING` breakdown on Z; in motion.cpp:

https://github.com/MarlinFirmware/Marlin/blob/8e06e9ca40ffad4a37b7f4a43dec36b63868b92f/Marlin/src/module/motion.cpp#L1796-L1800

So, the second homing probe is buried under the `bump` conditional, which is defined as needing a HOMING_BUMP_MM for the axis..

https://github.com/MarlinFirmware/Marlin/blob/8e06e9ca40ffad4a37b7f4a43dec36b63868b92f/Marlin/src/module/motion.cpp#L1780-L1783

https://github.com/MarlinFirmware/Marlin/blob/8e06e9ca40ffad4a37b7f4a43dec36b63868b92f/Marlin/src/module/motion.h#L136-L139

### Bug Timeline

After enabling sensorless homing, which recommends setting all homing bumps to 0 (and recommendations were followed).

### Expected behavior

I expected the second Z probe to be based on the `MULTIPLE_PROBING` and/or `EXTRA_PROBES` settings, thus I expected Z to double probe on homing.

### Actual behavior

With `HOMING_BUMP_MM` set to 0 for Z, `MULTIPLE_PROBING` is not respected for homing Z. There does not seem to be any documentation/comments that allude to this being the case.

### Steps to Reproduce

1. Have a Z probe, set `MULTIPLE_PROBING` to 2 (or more), set `HOMING_BUMP_MM` to `{0, 0, 0}`
2. `G28`
3. Probe only probes once.

### Version of Marlin Firmware

bugfix-2.0.x latest

### Printer model

"ender 3" abomination

### Electronics

SKR 1.4, TMC2209 drivers

### Add-ons

BLTouch

### Additional information & file uploads

I'm unsure what would be the best solution to this, but I have a few ideas:

1. Change `homeaxis(...)` to use `MULTIPLE_PROBING` value to detemine how many probes should be executed for Z, if `HOMING_Z_WITH_PROBE` is enabled.
2. Update comments/docs for `MULTIPLE_PROBING` to note it only applies to bed leveling/G38/etc.
3. Update comments/docs for `HOMING_BUMP_MM` to note that it controls multiple-probe on Z homing.

Either way, the comment `SENSORLESS_HOMING` should reflect only disabling `HOMING_BUMP_MM` for axes that you're enabling sensorless homing on.

slowbro, 2021-06-15:
Well, I just found #9802, 3+ years old! It seems to also talk about this issue, though from my perspective multi-probing for Z always "worked" since I have had `HOMING_BUMP_MM` enabled for as long as I can remember.

Not sure if this should be closed in favor of that, or not.

thisiskeithb, 2021-06-15:
> Well, I just found #9802, 3+ years old! It seems to also talk about this issue

Yes, please consolidate discussions into that issue.

thisiskeithb, 2021-06-15:
Duplicate of #9802

closed 2021-06-15 by thisiskeithb