MarlinFirmware/Marlin #22140: [BUG] Unable to get BTT Mini E3 v1.2 to connect to OctoPrint

Taomyn, 2021-06-15:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

I had not built a bugfix-2.0.x firmware since 16th May and so have been trying to get it to work for the past week. After discovering that the 512K build for the BTT Mini E3 v1.2 is now unsupported I have switched to the standard 256K build. However, no matter what I try my OctoPrint server will not connect to it over USB, it fails to find the serial-over-USB connection.

Reverting to my 16th May build fixes the problem.

### Bug Timeline

New

### Expected behavior

OctoPrint should connect

### Actual behavior

OctoPrint cannot find the printer

### Steps to Reproduce

Get latest bugfix-2.0.x, compile and load onto printer.

### Version of Marlin Firmware

bugfix-2.0.x

### Printer model

Creality Ender 3 Pro

### Electronics

BTT Mini E3 v1.2

### Your Slicer

Cura

### Host Software

OctoPrint

### Additional information & file uploads

I have tried removing the code to enable NeoPixels and also lowered the serial speed - neither has helped.

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/6653406/Configuration.zip)

ellensp, 2021-06-15:
Circuit diagram shows that this controllers USB port is directly connected to the controller CPU. So this is serial port -1
There is also a serial port as part of the TFT plug, this is serial port 2
Lastly on on the lcd connector is serial port 1 

The provided configuration files only has #define SERIAL_PORT 2, ie only the TFT plug has been ababled.

Your config also has //#define SERIAL_PORT_2 -1, so the USB port is disabled.

You need to use enable #define SERIAL_PORT_2 -1

Taomyn, 2021-06-15:
@ellensp thanks for the prompt reply. It does seem that a change since the 16th May I did not merge very well with my changes and the second port got disabled as you suspect. However I now cannot compile it to fit in 256KB - I really wish the decision to remove 512KB was done via a sanity check rather than completely being removed.

ellensp, 2021-06-15:
If your sure your 256-512k is ok (since most of them are not) or you have a chip labeled STM32F103RE and not the smaller RC version. you can change build env from STM32F103CE_btt to STM32F103RE_btt

in Marlin/Marlin/src/pins/pins.h is the lines
```
#elif MB(BTT_SKR_MINI_E3_V1_2)
  #include "stm32f1/pins_BTT_SKR_MINI_E3_V1_2.h"  // STM32F1                              env:STM32F103RC_btt env:STM32F103RC_btt_USB env:STM32F103RC_btt_maple env:STM32F103RC_btt_USB_maple

```
Add env:STM32F103RE_btt  to the list of valid environments and build with the environment STM32F103RE_btt

ellensp, 2021-06-15:
I did'nt like my previous answer, may not work..  but this will
If your sure your 256-512k works. You can just add it back if you want
https://github.com/MarlinFirmware/Marlin/pull/22050/files list all changes that removed it

ie add this to ini/stm32f1.ini 
```
[env:STM32F103RC_btt_512K]
platform          = ${common_stm32.platform}
extends           = env:STM32F103RC_btt
board_upload.maximum_size = 524288
build_flags       = ${env:STM32F103RC_btt.build_flags} -DLD_MAX_DATA_SIZE=524288 -DSTM32_FLASH_SIZE=512
```
and add  env:STM32F103RC_btt_512K to the list of valid environment for this board

ie add to the list of env's in pins.h
```
#elif MB(BTT_SKR_MINI_E3_V1_2)
  #include "stm32f1/pins_BTT_SKR_MINI_E3_V1_2.h"  // STM32F1                              env:STM32F103RC_btt_512K env:STM32F103RC_btt env:STM32F103RC_btt_USB env:STM32F103RC_btt_maple env:STM32F103RC_btt_USB_maple env:STM32F103RC_btt_512K

```

Taomyn, 2021-06-15:
@ellensp thanks again, it should be fine as I have been using the 512KB build since Jan 2020 and never had an issue that wasn't something I did wrong, or a bug.

I'll try it and let you know.

Taomyn, 2021-06-15:
@ellensp thank you again, that's done the trick - I will point my other thread on the 512KB to this one in case anyone else needs to get it back

closed 2021-06-15 by Taomyn

ellensp, 2021-06-15:
Reinstating my first reply. 

If your sure your 256-512k is ok (since most of them are not) or you have a chip labeled STM32F103RE and not the smaller RC version. you can change build env from STM32F103CE_btt to STM32F103RE_btt

in Marlin/Marlin/src/pins/pins.h is the lines
#elif MB(BTT_SKR_MINI_E3_V1_2)
  #include "stm32f1/pins_BTT_SKR_MINI_E3_V1_2.h"  // STM32F1                              env:STM32F103RC_btt env:STM32F103RC_btt_USB env:STM32F103RC_btt_maple env:STM32F103RC_btt_USB_maple

Add env:STM32F103RE_btt

then build with that environment.